omersiar/esp-rfid #1: nodemcu pin

akeilox, 2017-05-28:
im using the nodemcu (amica)  from https://www.banggood.com/NodeMcu-Lua-WIFI-Internet-Things-Development-Board-Based-ESP8266-CP2102-Wireless-Module-p-1097112.html?rmmds=search

i noticed the pin connection comments at top of source code, sda D2 & RST D1 , and later in the code different pins selected referring to ''wemos d1 mini''
// Configure RC522 Device Pins on
#define SS_PIN D8                     //Pin on WeMos D1 Mini
#define RST_PIN D3                    //Pin on WeMos D1 Mini

Are you testing this on WeMos D1 Mini, and if yes which version would that be?

akeilox, 2017-05-28:
I have tried adjusting the pins as per ss->d2, rst -> d1 in code, compile and uploaded.

I get the following output on serial but no card event shown on web interface, (if i did tried running without adjusting pins, the serial output was mfrc not connected properly ...
So after the two pins adjusted, the error message gone and shows firmware oxDE, and web interface shows up with bootstrap etc. loaded - but no card scan.

(to note, i have tested the same card and mfrc522 on arduino uno r3, with same mfrc522 library, so not sure what im missing here)

ESP RFID AccessControl v0.1
Connecting..............................
Couldn't connect in time
Configuring access point... Ready
AP IP address: 192.168.4.1
MDNS responder started
RFID Hardware Firmware Version: 0xDE = (unknown)
ws[/ws][1] connect
ws[/ws][1] disconnect: 0
ws[/ws][2] connect
ws[/ws][2] disconnect: 0
ws[/ws][3] connect

omersiar, 2017-05-28:
i believe each board have their explict pin number defines. look for spi pins of modemcu board, they are specific pins.

yes i use wemos d1 mini, regardles of used boards ss and rst pins are configurable.

i did not encounter 0xde version of rfid hardware, please check enough current flows to both esp and rfid hardware.

i will add more debugging code this should make troubleshooting more easy.

omersiar, 2017-05-31:
@akeilox Did you managed to get it work?

akeilox, 2017-05-31:
i get jibberrish symbols on console if i use the pins SS D8 and RST D3 (from original code).

Only when I connect SS_PIN to D2 and RST_PIN to D1 (other pins as per the default in the header),
i get a serial output as following, everything seems to work (http, file upload, connect, disconnect) except reading of mifare card on MFRC522;

ESP RFID AccessControl v0.1
Connecting..............................
Configuring access point... Ready
AP IP address: 192.168.4.1
MDNS responder started
RFID Hardware Firmware Version: 0xDE = (unknown)

i do use the very same mfrc522 module on arduino project and it works, so its not the module or cable, and seems only getting console output when using the default stated pins;
/*
   Typical pin layout used:
   ------------------------------------
               MFRC522
               Reader/PCD   ESP8266/
   Signal      Pin          Nodemcu
   ------------------------------------
   RST/Reset   RST          GPIO5/D1
   SPI SS      SDA(SS)      GPIO4/D2
   SPI MOSI    MOSI         GPIO13/D7
   SPI MISO    MISO         GPIO12/D6
   SPI SCK     SCK          GPIO14/D5
*/

I have attached the view from arduino ide, clarifying the version, and board and upload method for reference and benefit for others facing similar issue;
[image]

omersiar, 2017-05-31:
actually ss pin does not to be connected to any pin, and recent update to library makes possible to not use rst pin.

simply connect ss pin to gnd (of course this is only workaround will not work if there is another device on spi bus)

and try to power rfid module as like you do as on uno tie gnd s together

akeilox, 2017-06-01:
thanks for the headsup, i have downloaded the latest rfid-master from https://github.com/miguelbalboa/rfid
to update the arduino lib,

and connected SS pin to GND and did not connect the RST pin as you suggested

When I compile and upload, i get the following on the console;

Configuring access point... Ready
AP IP address: 192.168.4.1
MDNS responder started
RFID Hardware Firmware Version: 0xFF = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?

omersiar, 2017-06-01:
SDA > D4 (GPIO02), 
SCK > D5 (GPIO14), 
MOSI > D7 (GPIO13), 
MISO > D6 (GPIO12), 
GND > GND, 
RST > D4
3.3V > 3.3V

can you connect it to those pins and check?

omersiar, 2017-06-01:
MISO - GPIO12 (hw spi)
MOSI - GPIO13 (hw spi)
SCK  - GPIO14 (hw spi)
SS   - GPIO04 (free GPIO)
RST  - GPIO05 (free GPIO)

look for pinouts for your board

omersiar, 2017-06-01:
#include "MFRC522.h"
#define RST_PIN	5 // RST-PIN for RC522 - RFID - SPI - Modul GPIO5 
#define SS_PIN	4 // SDA-PIN for RC522 - RFID - SPI - Modul GPIO4 
MFRC522 mfrc522(SS_PIN, RST_PIN);	// Create MFRC522 instance

omersiar, 2017-06-04:
Any news? If you accomplish to connect, i would like to hear it, I will modify the readme for connections.

omersiar, 2017-06-05:
I just found out that the pins are same for both boards (WeMos and NodeMcu).

akeilox, 2017-06-14:
Apologies for the late reply. After numerous attempts I couldnt get it to work and ordered wemos mini, esp32 and spare 8286-12E and received the esp32 first while others on transit.
It became apparent that mfrc lib version 1.3.6 does not work with esp32 nor the nodemcu 12e
and switching the version to 1.3.5 makes it all work without any changes.

So to conclude the pin mapping for nodemcu, the following works on mfrc lib version 1.3.5
#define SS_PIN D8 //Pin on WeMos D1 Mini
#define RST_PIN D3 //Pin on WeMos D1 Mini

as well as on ESP32 and WeMos mini.

I can finally run the code on nodemcu 12e as its designed and loved the approach you have taken!

akeilox, 2017-06-15:
The pin mapping for nodemcu, the following works on mfrc lib version 1.3.5 but not in 1.3.6. Change version within Arduino IDE and all is good.

#define SS_PIN D8 //Pin on WeMos D1 Mini
#define RST_PIN D3 //Pin on WeMos D1 Mini

as well as on ESP32!

closed 2017-06-15 by akeilox