omersiar/esp-rfid #3: what would be the ideal way to communicate with a notify/mgmt server

akeilox, 2017-06-15:
Assuming communicating with a remote server is an option, in case its considered I have the following proof of concept that I run to see the time it takes between communication and test different scenarios.

This is purely to initiate discussion and not meant to be a polished integration;
Question/points it raises;

-  if httpclient is the right approach, or websocket based, async is the way to go for posting to remote server.

- SSL seems possible by declaring the ca certificate at the top (untrusted but works), which can also be added (trusted) via settings in the future.

- would it be possible to enforce 2 person entry; i.e. a UID needs to be scanned with another (paired) UID to grant access. Perhaps within 10 second time window. UID x is stored with **pair** field as the other UID y - thats how perhaps we can handle it simply, if pair exists wait for pair to be scanned. 

- it makes more sense to pass JSON to server, rather than querystring it seems. Im learning as I go along so any input are welcome.

```
#include <ESP8266HTTPClient.h>

String notifyserver="192.168.1.2";
String handler="/mgmt/index.php?";
String deviceid="MainDoor";
String hquery= "http://"+ notifyserver + handler + "device="+ deviceid + "&uid=";
// should pass the deviceid, uid, timestamp in JSON ideally, not in querystring.

void pingMgmtServer()
{

  HTTPClient http;
 Serial.print("[HTTP] begin...\n");
 http.begin(hquery);
 //if we want to do it with HTTPS, need ca certificate defined above
 //  http.begin("https://notifyserver/handler/?", ca); //HTTPS 
        
        Serial.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();
        
String payload="";

        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            Serial.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                 payload = http.getString();
                Serial.println(payload);
              
            }
            
             
                
        } else {
            Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();

 switch (payload) {
    case "logged":    // logged in server/db
      Serial.println("logged record.");
      break;
    case "removed":    // indicates uid access right is removed
      Serial.println("the uid has been removed, do we do something?");
      break;
    case "added":    // n/a
      Serial.println("add not implemented");
      break;
    case "twokey":    // this uid needs another uid to be scanned before/after; i.e. 2person presense entry
      // perhaps the paired (second) uid can be send by server to client within the payload.
      Serial.println("this card needs the 2nd pair to be tapped to allow entry");
      break;
  }

       

}
```

omersiar, 2017-06-15:
I just wanted to inform you about the other project i came across while looking for esp security.

https://jjssoftware.github.io/id8266-aka-skynet/

This is almost perfect solution for what I am trying to do in general. 

It sounds like too good to be true :)

bootstrap responsive design
user authentication / session support
flash persistence for all configuration
network configuration support
DNS support
web server configuration support
GPIO configuration and control
GPIO MQTT support for open lightweight monitoring and control
websockets support
JSON Rest style API for 3rd party application development
NTP client configuration support
Support for logging to external services e.g. Thingspeak and any other HTTP server that accepts a GET request
diagnostics display
OTA firmware update

omersiar, 2017-06-15:
This is the closest thing what you want to do, and more secure due to hashing.
https://github.com/ParalelniPolis/rfid-locks/blob/master/locks.ino

omersiar, 2017-08-07:
As I mentioned before, I want ESP-RFID project to become sitting alone, depending on nothing but itself. This way we are simplifying working scheme.

Advanced project that uses server/client architecture can be found here:

https://github.com/torehc/carontepass-v2

closed 2017-08-07 by omersiar