omersiar/esp-rfid #4: SPIFFS.begin(); Twice

farthinder, 2017-06-25:
This is my first time working with SPIFFS and my first time submitting an issue so please forgive me if I misunderstand something but you seam to run " SPIFFS.begin();" twice in your code.

I really enjoy this project, keep up the good work!

omersiar, 2017-06-25:
feedbacks are always welcomed. thank you, unused frangmented code is everywhere and im very slopy about that, but there will be a release that will be reviewed.

omersiar, 2017-07-12:
I tried my best to review the code, code quality tools for Javascript is pretty straight forward, but not for the C++ (Arduino). HTML part was also easy to validate.

I know the Arduino code itself not even near "OK", but it just works. I tried to make it optimized. v0.1 is ready, mostly reviewed (not from a peer, but from a machine).

Thank you for your interest for the project. Kind regards.

closed 2017-07-12 by omersiar