omersiar/esp-rfid #5: 'class ESP8266WiFiClass' has no member named 'scanNetworksAsync'

jeremyoha450, 2017-06-29:
Hi there

when i try to compile the code i get the below error. i have esp2866 2.3.0.rc3 installed and i still get this error.

Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
C:\Users\Jem\Documents\Arduino\esp-rfid-master\esp-rfid\esp-rfid.ino: In function 'void onWsEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, uint8_t*, size_t)':

esp-rfid:272: error: 'class ESP8266WiFiClass' has no member named 'scanNetworksAsync'

         WiFi.scanNetworksAsync(printScanResult);

              ^

exit status 1
'class ESP8266WiFiClass' has no member named 'scanNetworksAsync'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

farthinder, 2017-06-29:
Hi!

   - Currently only Git version (2.4.0rc) of ESP8266 Core is supported, due
   to new function is introduced (WiFi.scanNetworksAsync())

You need a newer version of esp8266 core. It has to be downloaded from git.
You will likely get two boards with the same name in the Arduino ide after
installing this. If you are still having problems try switching to the
duplicate board.

jeremyoha450, 2017-06-29:
Thanks for you reply. i found 2.4.0.rc1 and installed and uploaded no problems

problem i;m having now is when it boots i get this
[ INFO ] ESP RFID v0.0.3
[ INFO ] Chip ID: 0x136DDF
[ INFO ] Hostname: rfid-door136ddf
[ WARN ] Failed to open config file
[ INFO ] Configuring access point... Ready
[ INFO ] AP IP address: 192.168.4.1

i have tried accessing http://192.168.4.1 and http://192.168.4.1/ws and http://192.168.4.1/admin and http://192.168.4.1/auth

and i just get cant find page

regards
Jeremy

farthinder, 2017-06-29:
You need to upload the data-folder using ESP8266FS Uploader. It's not the best software and tricky to use unfortunately. 

Check out the read me file :)

jeremyoha450, 2017-06-29:
Thanks for that

I did read it and download everything. but i didnt know i had to use the ESP8266FS Uploader as well. still learning arduino.

no steps saying upload sketch and then run ESP8266FS Uploader to upload file to complete the process.

regards
Jeremy

farthinder, 2017-06-29:
Inspired by and thanks to this project im creating a simpler version without a fancy web interface which will likely integrate with blynk and/or home assistant. I pretty much want to let the Arduino handle all the authentication and adding/removing rfid tags locally and use blynk/HA for logging and notifications.

jeremyoha450, 2017-06-29:
Are you going to add in MQTT? and add ability to add a door sensor (show its open) or a button (door bell)?

overall its good. i found a little glitch with logging in with Firefox. but other then that its look good. keep up the good work :)

jeremyoha450, 2017-06-29:
dont you know about this. but if you change the Antenna Gain and save it. it reverts back to avg(33 db)

farthinder, 2017-06-29:
I want to be clear about I am not a developer for this project, just somebody with to much time on my hands :) 
I´ve plaid around a little bit with changing the gain without and measurable difference.

jeremyoha450, 2017-06-29:
thats cool. i just noticed it thats all. i'll start to tweak it my self :) i'll just have to remember what i did for any updates you do :)

omersiar, 2017-06-30:
Whole ESP-RFID project is still in its infancy. It will mature in time eventually. Currently it serves everything for tinkerers, Readme still needs more information.

Antenna Gain setting is just dirty hack, but it works in my setup, and there is a huge difference between Min and Max in my case.

Logging In to settings page will be changed, more likely become a cookie based authentication.

@farthinder Also there is this code for AVR boards: 
https://github.com/omersiar/RFID522-Door-Unlock

@jeremyoha450 Please do submit your findings (bugs, glitch, suggestions, etc).

Kind regards.

farthinder, 2017-06-30:
Thanks @omersiar! I´m shooting for something in-between your project and the one you mentioned, but great to have one more project for inspiration!

omersiar, 2017-08-07:
I left open this issue because new comers may have encounter this issue as well. Readme updated and this issue can be closed now. Thank you.

closed 2017-08-07 by omersiar