omersiar/esp-rfid #6: Reset-pin no longer needed

farthinder, 2017-07-12:
I needed more pins on my WeemosD1 and set out to free up some, discovered that the Reset-pin for the RFID-reader is no longer needed as it´s now done in software:

https://github.com/miguelbalboa/rfid/pull/275/files

omersiar, 2017-07-12:
I was aware of that, but did not test it through. Made changes, and tested it. Thanks for the heads up. :+1:

closed 2017-07-12 by omersiar

farthinder, 2017-07-12:
Nice!

Note quite sure how to reference code in an issue, but this should perhaps also be updated?
`

The following table shows the typical pin layout used:
--
  |  
  | | Signal        | MFRC522       | WeMos D1 mini  | NodeMcu | Generic      |
  | |---------------|:-------------:|:--------------:| :------:|:------------:|
  | | RST/Reset     | RST           | D3 [1]         | D3 [1]  | GPIO-0 [1]   |

`

omersiar, 2017-07-12:
Yes, it's updated now. Thank you.

khumprp, 2022-06-15:
Sorry for reviving the dead - I'm using a TTGO TDISPLAY where the RST pin is not made available in pinout. It's used to control the TFT I believe.

I'm trying to figure out how to bypass the need for the RST pin. Here's my code:

MFRC522 mfrc522(SS_PIN, UINT8_MAX);  // Create MFRC522 instance

Unfortunately, no luck.

The RC522 works when I hook it to a different ESP32 where the RST is available.

Any help would be greatly appreciated!