omersiar/esp-rfid #7: Not an issue but an implementation

rneurink, 2017-07-19:
Not an issue but I've been working on my own RFID system and i thought lets merge these projects. So heres some code to implement a timing system

The timing system makes it available so someone can select times for each day to allow someone to enter.
[image]

I'm still working on converting my logging system to this project. Feel free to ask questions

users.htm:
`<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="utf-8">
    <meta http-equiv="X-UA-Compatible" content="IE=edge">
    <meta name="viewport" content="width=device-width, initial-scale=1">
    <meta name="description" content="">
    <meta name="author" content="">
    <link rel="icon" href="/favicon.ico">
    <title>ESP RFID</title>
    
    <link href="/bootstrap.css" rel="stylesheet">
    <script src="/auth/users.js" type="text/javascript" charset="utf-8"></script>
</head>
<body onload="start()">
    <div class="container">
        <div class="header clearfix">
            <nav>
                <ul class="nav nav-pills pull-right">
                    <li role="presentation" class="active"><a href="/auth/users.htm">Users</a> </li>
                    <li role="presentation"><a href="/auth/settings.htm">Settings</a> </li>
                    <li role="presentation"><a href="/auth/log.htm">Log</a></li>
                </ul>
            </nav>
            <h3 class="text-muted">ESP RFID</h3>
        </div>
        <div class="jumbotron">
            <div class="row">
                <div class="col-sm-6 col-sm-push-6">
                    <legend>User List</legend>
                    <div class="overlay">
                        <div id="loading-img" class="text-center">
                            [image]
                            <h5>Please wait while WebSocket is connecting...</h5>
                            <br>
                        </div>
                    </div>
                    <div class="panel panel-default table-responsive">
                        <table id="knowntable" class="table table-hover table-striped">
                            <thead class="thead-default">
                                <tr>
                                    <th>UID</th>
                                    <th>Label / User Name</th>
                                    <th>Access</th>
									<th style="display:none"></th>
                                </tr>
                            </thead>
                            <tbody id="tablebody"></tbody>
                        </table>
                    </div>
                </div>
                <div class="col-sm-6 col-sm-pull-6">
                    <legend>Add / Update User</legend>
                    <p style="font-size:12px;" class="text-muted">Scan a PICC (RFID Tag) or choose from the list.</p>
                    <form id="fade">
                        <div class="input-group">
                            <span class="input-group-addon" style="text-align:left;min-width:100px"><strong>UID</strong></span>
                            <input id="uidinp" type="text" style="max-width:290px" class="form-control" name="msg" placeholder="PICC UID" disabled>
                        </div>
                        <div class="input-group">
                            <span class="input-group-addon" style="text-align:left;min-width:100px"><strong>PICC Type</strong></span>
                            <input id="typeinp" type="text" style="max-width:290px" class="form-control" name="msg" placeholder="PICC Type" disabled>
[... 1296 lines cut ...]
	{}
}

/* ------------------ Misc Functions ------------------- */
void turnOnLed(int pin) {
	switch (pin) {
	    case greedLed:
	    	digitalWrite(redLed, LOW);
	    	digitalWrite(blueLed, LOW);
	      	break;
	    case redLed:
		    digitalWrite(redLed, HIGH);
	    	digitalWrite(blueLed, LOW);
	      	break;
	    case blueLed:
		    digitalWrite(redLed, LOW);
	    	digitalWrite(blueLed, HIGH);
	      	break;
	    default:
	    	break;
	}
}
`

[RFID_ESP.zip](https://github.com/omersiar/esp-rfid/files/1159353/RFID_ESP.zip)

omersiar, 2017-07-27:
- [ ] Add Captive Portal 
- [ ] Add Timed entry

from @rneurink 's fork

- [ ] Add ability to choose administrative password
- [ ] Add ability to choose time zone, NTP Server, Browser Time Sync (for AP Mode)

What else for v0.2?

akeilox, 2017-07-27:
This project is coming along very nicely since I last checked, loving the direction being taken!

Are they any plans to change the SS and RST pin selection (esp if rst is redundant now) in GUI to Enter and Exit pin selection so it can support 2 mfrc522 units connected to the same mcu? Perhaps the default is Enter and Exit being optional.
So one can do logs like;

Name BOB Card A1235B Entered [Door-Name] at [Date and Time]
...
Name BOB Card A1235B Exited [Door-Name] at [Date and Time]

omersiar, 2017-07-27:
Hello @akeilox, what you are asking is already in my to do list, but it may take a long time to get there, maybe v0.4? because i do not have an another RC522 at the moment and there are other things that really really need to be done before this.

akeilox, 2017-07-27:
Hi @omersiar understand, I been side tracked now catching up from stored email notifications and code updates, this is a great learning tool with very cool practical applications.

I saw the progress on SPIFF, and was wondering if you will consider adding support for paired-access.
Say like for a store room or file room, where two cards (person) need to be tapped to gain access.
When cardid stored in EPROM, perhaps it can have an optional pair field (name,cardid,pairid... ), and if this field is filled with another card id then mcu will await that card to be scanned too, to give access.
Is it something that can be factored in to the design for future implementation.
I am interested to hear your thoughts on this.

omersiar, 2017-07-27:
Yeah totaly esp-rfid can work like this, you can divide users into two groups, one have direct access, one have to wait for another person to validate access, actually you can do it now, for second group, you do not even have to deal with eeprom or know them. Just give users rfid tags, and do not let them access, give security guy the valid tag, lets people to enter

omersiar, 2017-07-27:
i assume there is a security guy, but if there are multiple parties for access, i do not know how to actually maintain this. design your work flow, perfect it, then coding will be easy.

rneurink, 2017-08-01:
> Are they any plans to change the SS and RST pin selection (esp if rst is redundant now) in GUI to Enter and Exit pin selection so it can support 2 mfrc522 units connected to the same mcu? Perhaps the default is Enter and Exit being optional.
So one can do logs like;

I've got 2 readers so i will work on this after i get the SD card with logging working.

omersiar, 2017-08-01:
This was done there nicely: https://github.com/torehc/carontepass-v2/blob/master/arduino/caronte-esp8266/caronte-esp8266-two-rfid.ino

omersiar, 2017-08-01:
@rneurink Also I am still trying to convince my self about timed entry.

I am thinking about global Open-Close hours in Settings Page 

https://codepen.io/jasesmith/pen/reLEyb

and also beside from that setting, every user can also be individually configured as you did it. I need somehow more easy approach for Timed Access, for example selecting boxes for hours of day and day of week like they are doing with torrent clients. 

https://www.raymond.cc/blog/schedule-bittorrent-to-automatically-start-and-stop-downloading/

Also for future reference MomentJS is very promising: https://momentjs.com/

omersiar, 2017-08-07:
Timed Access (Entry) will likely be in V0.4

closed 2017-08-07 by omersiar