omersiar/esp-rfid #10: v0.2

omersiar, 2017-08-05:
- [ ] Get SSID, connect to BSSID
- [ ] pseudo code for give access until specific date.
`if (epoch <= validuntil) { grantaccess() }
- [ ] fix status command in AP mode
- [ ] add "available/used space on SPIFFS" to status

omersiar, 2017-08-08:
- [ ] Add support for active low relays

rneurink, 2017-08-08:
I've fixed the status command in AP mode. The only thing i couldn't find is how to get the dns server when in AP mode

`
#ifdef ESP8266
extern "C" {
#include "user_interface.h"		// Used to get Wifi status information
}
#endif

void sendStatus() {
	DynamicJsonBuffer jsonBuffer;
	JsonObject& root = jsonBuffer.createObject();
	int mode = WiFi.getMode(); //const char* modes[] = { "NULL", "STA", "AP", "STA+AP" };

	struct ip_info info;

	if (mode == 1) { //Station mode
		wifi_get_ip_info(STATION_IF, &info);
		struct station_config conf;
		wifi_station_get_config(&conf);
		root["ssid"] = String(reinterpret_cast<char*>(conf.ssid));
		root["dns"] = printIP(WiFi.dnsIP());
		root["mac"] = WiFi.macAddress();
	} else if (mode == 2) { //SoftAP mode
		wifi_get_ip_info(SOFTAP_IF, &info);
		struct softap_config conf;
		wifi_softap_get_config(&conf);
		root["ssid"] = String(reinterpret_cast<char*>(conf.ssid));
		root["dns"] = "";
		root["mac"] = WiFi.softAPmacAddress();
	}
	IPAddress ipaddr = IPAddress(info.ip.addr);
	IPAddress gwaddr = IPAddress(info.gw.addr);
	IPAddress nmaddr = IPAddress(info.netmask.addr);
	root["ip"] = printIP(ipaddr);
	root["gateway"] = printIP(gwaddr);
	root["netmask"] = printIP(nmaddr);

	root["command"] = "status";
	root["heap"] = ESP.getFreeHeap();
	root["chipid"] = String(ESP.getChipId(), HEX);
	root["cpu"] = ESP.getCpuFreqMHz();
	root["availsize"] = ESP.getFreeSketchSpace();
	
	size_t len = root.measureLength();
	AsyncWebSocketMessageBuffer * buffer = ws.makeBuffer(len); //  creates a buffer (len + 1) for you.
	if (buffer) {
		root.printTo((char *)buffer->get(), len + 1);
		ws.textAll(buffer);
	}
}
`

rneurink, 2017-08-08:
give access untill specific date is also included in my fork [Commit](https://github.com/rneurink/esp-rfid/commit/41bc865142fdae069c3e8b5ff7a88342d89d0516)

rneurink, 2017-08-08:
added visualisation to see how much spiffs space is available [commit](https://github.com/rneurink/esp-rfid/commit/5b962538bbf1c3234c05cea9ec8bf24f81ad6561)

I was thinking is it not more logical to display space used instead of space available?

omersiar, 2017-08-08:
Nicely done thank you. I even did not understand how ESP8266 memory mapped. Heap and Flash calculations are just my assumptions based on Arduino IDE compile time feedback.

Most development boards have 16Mbit (4MB) flash. I believe we can get info about SPIFFS then make calculations.

```
struct FSInfo {
    size_t totalBytes;
    size_t usedBytes;
```

http://esp8266.github.io/Arduino/versions/2.0.0/doc/filesystem.html#info

I am stuck on Time. Timezones, UTC, GMT, conversions one from another, Day Light Saving (Summer Time) all these stuff making me pull my hair. I think i am going to give up on timing for now, and focus on fun stuff.

rneurink, 2017-08-08:
I've got the time from the ntpclient for now and it works fine. i've got a variable in which the UTC timezone can be set and it will adjust. 
For the SPIFFS mapping i am indeed using FSInfo. this is however an estimate and is not perfectly correct but it is good enough for this use.

I am currently cleaning and testing my code. I though it might be usefull to come up with a general layout for the .ini file so it's easier to add and remove code.

rneurink, 2017-08-08:
The memory is not that hard to understand. I'm using the esp12 which has 4M memory
the first 1M is used for the .ini file (flash)
the other 3M is used for SPIFFS
The way you get the remaining flash space is good so nothing to worry about here. I'm not sure yet if "1044464" in the set.js file is correct though. There might be a way to check this in runtime

omersiar, 2017-08-08:
I did not mean flash memory but random access memory.

https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map

Just finished my NTP code, too both Javascript and ino. https://github.com/gmag11/NtpClient using this library now.

It's an overkill but at least does not throw exceptions as TimeLib

omersiar, 2017-08-09:
@rneurink Thank you, it means a lot. :+1:

Also shifted some goals to v0.4. This issue now can be closed.

closed 2017-08-09 by omersiar