omersiar/esp-rfid #12: v0.4 Goals

omersiar, 2017-08-09:
- [ ] Add support for active low relays
- [ ] Tidy up everything

LifeP, 2017-08-29:
@omersiar Just stumbled upon this project while planning to do something similar. Very beautifully done! A couple of points I was thinking differently about -  
1. What do you think about using MQTT ( I think this could give better scalability and reliability. There are many projects using this with the wemosd1 mini itself)
2. How about storing card data in database on local web server rather than on flash.

Would love to hear what you think about them.

omersiar, 2017-08-29:
esp-rfid project's main goal is stand-alone access control system that depends on nothing but itself and ease of use. 

there are other projects that are based on client / server scheme.

even with 4mb flash we achieved 1.000 user goal mqtt support would be nice. maybe someone will implement general purpose use of it.

https://github.com/torehc/carontepass-v2

nardev, 2017-08-31:
I already did that with mqtt but with wiegand readers.

Unfortunately, in real life, it's not that applicable and important at all. Only good thing is ability to control multiple devices from same server.

Things like trigger OTA  upgrade, fast update  user lists, change rights for certain doors, follow user through building .... but rarely anyone need that...There are already systems of that kind 20+ years old. Doing all that...

closed 2018-01-18 by omersiar