omersiar/esp-rfid #13: NodeMCU has encountered a problem

1330154737, 2017-08-14:
Thank you for your project When I give the NodeMCU brush into the firmware, log on 192.168.4.1, in the interface after the operation click the save button is not used, the serial monitor will keep the code in the loop     a little advice: you can add a device to the 1602 iic screen， GPIO16 and GPIO2 are free 

- [ INFO ] ESP RFID v0.2
[ INFO ] Config file found
{
  "command": "configfile",
  "bssid": 0,
  "ssid": "HZDY",
  "wmode": "1",
  "pswd": "19970727",
  "sspin": "15",
  "rfidgain": "32",
  "rpin": "4",
  "rtime": "",
  "ntpserver": "",
  "ntpinterval": "",
  "timezone": "-12",
  "hostnm": "",
  "adminpwd": "admin"
}

Soft WDT reset

ctx: cont 
sp: 3fff11f0 end: 3fff1500 offset: 01b0

>>>stack>>>
3fff13a0:  00241001 00000001 3fff04e0 40106449  
3fff13b0:  00000002 3fff00e4 3fff01a4 4020e0b7  
3fff13c0:  402012fa 00000001 00000032 402012ef  
3fff13d0:  00000020 00000010 3fff00e4 4020e229  
3fff13e0:  00000001 0000000f 3fff00e4 4020e31d  
3fff13f0:  00000020 00000000 3fff00e4 4020e33b  
3fff1400:  00000000 0000000f 3fff3840 40207497  
3fff1410:  00000001 3fff0414 3fff3840 4020a337  
3fff1420:  3ffe9fb8 00000000 000003e8 40218d11  
3fff1430:  3fff36bc 3fff36fc 3ffe8dd8 00000000  
3fff1440:  3fff3834 00000200 4021907d 3fff145c  
3fff1450:  00100000 3fff1448 00000119 3fff0414  
3fff1460:  00000220 3fff3714 3fff0414 40216cfd  
3fff1470:  3ffe8b6c 00000000 3fff0414 3fff04d4  
3fff1480:  3fff1524 3fff2898 3fff2894 40218d5b  
3fff1490:  3fffdad0 00000000 3fff0414 3fff04d4  
3fff14a0:  3fff1524 00000000 3fff0414 4020a5ac  
3fff14b0:  feefeffe feefeffe feefeffe feefeffe  
3fff14c0:  feefeffe feefeffe feefeffe feefeffe  
3fff14d0:  feefeffe feefeffe feefeffe feefeffe  
3fff14e0:  3fffdad0 00000000 3fff04cd 40218164  
3fff14f0:  feefeffe feefeffe 3fff04e0 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vf6d232f1
~ld

omersiar, 2017-08-14:
please fill all the options in Settings page and save and reboot. Then it should be fine.

omersiar, 2017-08-14:
flash web files again to escape from boot loop

1330154737, 2017-08-14:
thank you very much , let me try

1330154737, 2017-08-15:
I re-brush into the firmware, according to the demo inside the image fill all the options in the Settings page, save and restart or no effect, serial monitor inside or loop display code, I use another esp8266 module is the case, you have Did other equipment test? Will it be caused by different types of esp8266 chips

omersiar, 2017-08-15:
how do you configure settings, can you paste here serial output.

1330154737, 2017-08-15:
please Wait a moment

1330154737, 2017-08-15:
[ INFO ] ESP RFID v0.2
[ INFO ] Config file found
{
  "command": "configfile",
  "bssid": 0,
  "ssid": "jun",
  "wmode": "1",
  "pswd": "12345678",
  "sspin": "15",
  "rfidgain": "32",
  "rpin": "4",
  "rtime": "250",
  "ntpserver": "europe.pool.ntp.org",
  "ntpinterval": "10",
  "timezone": "-12",
  "hostnm": "esp-rfid",
  "adminpwd": "admin"
}

Soft WDT reset

ctx: cont 
sp: 3fff1200 end: 3fff1500 offset: 01b0

>>>stack>>>
3fff13b0:  00000002 3fff00e4 3fff01a4 4020e09a  
3fff13c0:  402012fa 00000001 00000032 402012ef  
3fff13d0:  00000020 00000010 3fff00e4 4020e229  
3fff13e0:  00000001 0000000f 3fff00e4 4020e31d  
3fff13f0:  00000020 00000000 3fff00e4 4020e33b  
3fff1400:  00000000 0000000f 3fff33e8 40207497  
3fff1410:  00000001 3fff0414 3fff33e8 4020a337  
3fff1420:  3ffe9fb8 00000000 000003e8 40218d11  
3fff1430:  3fff3244 3fff3284 3ffe8dd8 00000000  
3fff1440:  3fff33dc 00000200 4021907d 3fff145c  
3fff1450:  00100000 3fff1448 00000138 3fff0414  
3fff1460:  00000220 3fff329c 3fff0414 40216cfd  
3fff1470:  3ffe8b6c 00000000 3fff0414 3fff04d4  
3fff1480:  3fff1524 3fff2420 3fff241c 40218d5b  
3fff1490:  3fffdad0 00000000 3fff0414 3fff04d4  
3fff14a0:  3fff1524 00000000 3fff0414 4020a5ac  
3fff14b0:  feefeffe feefeffe feefeffe feefeffe  
3fff14c0:  feefeffe feefeffe feefeffe feefeffe  
3fff14d0:  feefeffe feefeffe feefeffe feefeffe  
3fff14e0:  3fffdad0 00000000 3fff04cd 40218164  
3fff14f0:  feefeffe feefeffe 3fff04e0 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vf6d232f1
~ld

omersiar, 2017-08-15:
Did you connect your RFID Reader to your ESP? Is it a MFRC522 compatible module? Are you building your firmware from the source? or using compiled binaries?

omersiar, 2017-08-15:
Try to flash this firmware

[esp-rfidfix13.zip](https://github.com/omersiar/esp-rfid/files/1224422/esp-rfidfix13.zip)

1330154737, 2017-08-15:
I did not connect MFRC522 to esp, I use the compiledbin folder in the document, burn the address were 0x00000000 and 0x00100000

omersiar, 2017-08-15:
It's a known issue that if you do not connect to RFID hardware to your ESP, it may cause this problem, especially when you do first flash to your ESP. Try to connect individual SPI pins to 3.3V with 10k resistor, this may help. I will try to figure out what causes this issue. Try to completely power of your ESP after flashing, wait 30 seconds and power on it.

1330154737, 2017-08-15:
sorry, the original is not connected mfrc522 reason, really disturb you

1330154737, 2017-08-15:
[ INFO ] RFID SS_PIN: 15 and Gain Factor: 32
[ INFO ] MFRC522 Version: 0xF7 (unknown)
[ INFO ] ESP-RFID is running in AP Mode 
[ INFO ] Configuring access point... Ready
[ INFO ] AP IP address: 192.168.4.1
[ INFO ] AP SSID: jun
Okay now, thanks for your excellent project

omersiar, 2017-08-15:
Great, please review the ESP-RFID project, and come back and tell us what you think. What can be improved, how are you going to use it, etc. Thank you for your interest, kind regards.

closed 2017-08-15 by omersiar