omersiar/esp-rfid #14: esptool upload problem

turuwhiska, 2017-08-22:
hi i used esptool to upload de file .bin . the file latest.bin work fine, start the ap and conect to my network, but dont have the spiffs file system. i used:
./esptool.py  --port /dev/ttyUSB0 --baud 115200 write_flash -u -fm=dio  0x6B000 latestspiffs.bin i tested 0x00000 
0x40000
but dont work

omersiar, 2017-08-22:
Which ESP module do you have?

turuwhiska, 2017-08-22:
Esp07 and esp12-f

turuwhiska, 2017-08-22:
Esp07 and 12-f

turuwhiska, 2017-08-23:
the first message is A fatal error occurred: Timed out waiting for packet header
but i added -u option and upload fine but dont work
latest.bin work fine its posible charge the spiffs file system from this firmware?
need upload the two .bin or only need the latestspiffs.bin and latest.bin to the complet system?

omersiar, 2017-08-23:
Install Arduino IDE ESP8266 Core and FSUploader no need to install libraries. Use FSUploader for easiest approach. Also figure out which address write SPIFFS. Also you can use esp-rfid/edit HTML editor to upload files by hand.

omersiar, 2017-08-23:
[v0.3alpha.zip](https://github.com/omersiar/esp-rfid/files/1244438/v0.3alpha.zip)

turuwhiska, 2017-08-23:
i tested v0.3alpha but is the same
File Not Found

URI: /
Method: GET
Arguments: 0
dont have the web system . i supouse which this system is in the spiffs.bin no?
the spiffs.bin is upload en the address 0x00000 0x40000 or 12000?

omersiar, 2017-08-24:
[SPIFFS] address: 0x100000

turuwhiska, 2017-08-24:
when upload the file i have the same problem A fatal error occurred: Timed out waiting for packet header
if add the option -u dont have the problem 
minicom:
ESP RFID v0.3alpha
[ WARN ] Failed to open config file
[ INFO ] Configuring access point... Ready
[ INFO ] AP IP address: 192.168.4.1

and the problem now is the config file i think this is for the spiffs system
ctx: cont                                                                       
sp: 3fff12b0 end: 3fff15c0 offset: 01b0                                         
                                                                                
>>>stack>>>                                                                     
3fff1460:  00000002 3fff01a4 3fff0264 4020c547                                  
3fff1470:  00241001 00000001 3fff05a0 40106449                                  
3fff1480:  00000002 3fff01a4 3fff0264 4020e196                                  
3fff1490:  402012fa 00000001 00000032 402012ef                                  
3fff14a0:  00000020 00000010 3fff01a4 4020e325                                  
3fff14b0:  00000001 0000000f 3fff01a4 4020e419                                  
3fff14c0:  00000020 00000000 3fff01a4 4020e437                                  
3fff14d0:  00000000 0000000f 3fff3480 40207497                                  
3fff14e0:  00000001 3fff04d4 3fff3480 4020a47f                                  
3fff14f0:  3ffea098 00000000 000003e8 00000000                                  
3fff1500:  3fff3304 3fff3344 3ffe8ea8 002fb000                                  
3fff1510:  3fff3474 00000200 0000207d 3fff152c                                  
3fff1520:  00000200 3fff1518 00000111 3fff04d4                                  
3fff1530:  3ffe8c20 3fff335c 3fff04d4 3fff0594                                  
3fff1540:  3fff15e4 3fff24e0 3fff24dc 402190ff                                  
3fff1550:  3fffdad0 00000000 3fff04d4 3fff0594                                  
3fff1560:  3fff15e4 00000000 3fff04d4 4020a6b0                                  
3fff1570:  feefeffe feefeffe feefeffe feefeffe                                  
3fff1580:  feefeffe feefeffe feefeffe feefeffe                                  
3fff1590:  feefeffe feefeffe feefeffe feefeffe                                  
3fff15a0:  3fffdad0 00000000 3fff058d 40218508                                  
3fff15b0:  feefeffe feefeffe 3fff05a0 4010070c                                  
<<<stack<<<                                                                     
                                                                                
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)                                   
                                                                                
load 0x4010f000, len 1384, room 16                                              
tail 8                                                                          
chksum 0x2d                                                                     
csum 0x2d                                                                       
vf6d232f1                                                                       
~ld                                                                             
                                                                                
[ INFO ] ESP RFID v0.3alpha                                                     
[ INFO ] Config file found                                                      
{                                                                               
  "command": "configfile",                                                      
  "bssid": "",                                                                  
  "ssid": "",                                                                   
  "wmode": "0",                                                                 
  "pswd": "",                                                                   
  "sspin": "15",                                                                
  "rfidgain": "32",                                                             
  "rpin": "4",                                                                  
  "rtime": "",                                                                  
  "ntpserver": "",                                                              
  "ntpinterval": "",                                                            
  "timezone": "-12",                                                            
  "hostnm": "",

omersiar, 2017-08-24:
Re-flash SPIFFS and configure all your settings before clicking Save and Reboot button.

turuwhiska, 2017-08-24:
constant reset
Soft WDT reset                                                                  
                                                                                
ctx: cont                                                                       
sp: 3fff12c0 end: 3fff15c0 offset: 01b0                                         
                                                                                
>>>stack>>>                                                                     
3fff1470:  00241001 00000001 3fff05a0 40106449                                  
3fff1480:  00000002 3fff01a4 3fff0264 4020e1ac                                  
3fff1490:  402012fa 00000001 00000032 402012ef                                  
3fff14a0:  00000060 00000010 3fff01a4 4020e325                                  
3fff14b0:  00000001 00000000 3fff01a4 4020e419                                  
3fff14c0:  00000060 00000000 3fff01a4 4020e437                                  
3fff14d0:  00000000 00000000 3fff3820 40207497                                  
3fff14e0:  00000001 3fff04d4 3fff3820 4020a47f                                  
3fff14f0:  3ffea098 00000000 000003e8 00000000                                  
3fff1500:  3fff3684 3fff36c4 3ffe8ea8 002fb000                                  
3fff1510:  3fff3814 00000200 0000207d 3fff152c                                  
3fff1520:  00000200 3fff1518 00000131 3fff04d4                                  
3fff1530:  3ffe8c20 3fff36dc 3fff04d4 3fff0594                                  
3fff1540:  3fff15e4 3fff2870 3fff286c 402190ff                                  
3fff1550:  3fffdad0 00000000 3fff04d4 3fff0594                                  
3fff1560:  3fff15e4 00000000 3fff04d4 4020a6b0                                  
3fff1570:  feefeffe feefeffe feefeffe feefeffe                                  
3fff1580:  feefeffe feefeffe feefeffe feefeffe                                  
3fff1590:  feefeffe feefeffe feefeffe feefeffe                                  
3fff15a0:  3fffdad0 00000000 3fff058d 40218508                                  
3fff15b0:  feefeffe feefeffe 3fff05a0 4010070c                                  
<<<stack<<<                                                                     
                                                                                
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)                                   
                                                                                
load 0x4010f000, len 1384, room 16                                              
tail 8                                                                          
chksum 0x2d                                                                     
csum 0x2d                                                                       
vf6d232f1                                                                       
~ld                                                                             
                                                                                
[ INFO ] ESP RFID v0.3alpha                                                     
[ INFO ] Config file found                                                      
{                                                                               
  "command": "configfile",                                                      
  "bssid": 0,                                                                   
  "ssid": "Turuxxxxx",                                                         
  "wmode": "1",                                                                 
  "pswd": "xxxxx",                                                           
  "sspin": "0",                                                                 
  "rfidgain": "96",                                                             
  "rpin": "2",                                                                  
  "rtime": "1",                                                                 
  "ntpserver": "pool.ntp.org",                                                  
  "ntpinterval": "1",                                                           
  "timezone": "-12",                                                            
  "hostnm": "kk",                                                               
  "adminpwd": "xxxxx"                                                        
}

omersiar, 2017-08-24:
Does MFRC522 hardware is connected ? Please connect it and restart #13

turuwhiska, 2017-08-24:
yes the problem did the mfrc522 not connected now work!!!!!! thanks so much
other inssue....: 1)the web system have record a register forentry system? 
2) the pin for relee its always on

omersiar, 2017-08-25:
1) We do not have recording for now, but there will be a logging system (simple) in the next release.
2) There will be a setting for active low or active high relays in the next release. For now you need to change it in source.

turuwhiska, 2017-08-26:
thanks so much greatings from argentina!!!

closed 2017-08-26 by turuwhiska