omersiar/esp-rfid #15: error "Filesystem full" when uploading SPIFFS.bin

MrMontesa, 2017-09-02:
Hey,
I'm using a Wemos D1 mini trying to install your great tool. I'm not sure if I'm doing something wrong or the storage on the wemos is simply too small.
I flashed the sketch and can properly connect to the wifi. But when flashing the SPIFFS via esp8266fs.jar I do receive the following error message:

```
[SPIFFS] data   : C:\Users\sebastian\Documents\Arduino\SPIFFS_upload\data
[SPIFFS] size   : 3052
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/latestspiffs.bin

SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!
```
Do i have to clear the FS before attempting to re-flash SPIFFS? Or is the flash too small?
Thanks for your help

omersiar, 2017-09-02:
Hi,

i do not know what causes this error, how are you creating spiffs binary, clearly it is not the way how you flash web files.

just have the data folder and do not copy anything to there.

then use uploader tool.

another way around is using built in html editor to upload files. you can access it via your browser goto http://ipaddress/edit

MrMontesa, 2017-09-02:
Thanks for the swift reply. I used your pre compiled bin file and uploaded it via arduino ide extension. I thought that would be the recommend way. When not placing anything in the data folder I got error like: no file found.
Will give it another try. As well as your suggestion with the edit link.
Thanks much.

omersiar, 2017-09-02:
then your data folder is empty either download fresh copy of the project with download option on github and use spiffs upload tool or flash latestspiffs.bin with esptool. they are different aproaches since you have spiffs upload tool, i recommend using of it.

data folder should like this

index.htm
logo.gif
required.cs
.
.
auth folder and its content

MrMontesa, 2017-09-02:
just gave it a try with the edit link and can upload files.
Unfortunately I cant create the auth & fonts folder via that GUI. Will give the Arduino GUI another try and clone the project.

MrMontesa, 2017-09-02:
Tried the Arduino GUI upload once more and failed with "cant connect to COM5", but my serial monitor is connected properly on that port and updates as well.
Here's the error message I get:

```
[SPIFFS] data   : C:\Users\sebastian\Documents\Arduino\esp-rfid-master\esp-rfid\data
[SPIFFS] size   : 3052
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/auth/log.htm
/auth/log.js
/auth/set.js
/auth/settings.htm
/auth/users.htm
/auth/users.js
/bootstrap.css.gz
/favicon.ico
/fonts/glyph.eot
/fonts/glyph.svg
/fonts/glyph.ttf
/fonts/glyph.woff
/fonts/glyph.woff2
/index.htm
/load.gif
/required.css.gz
/required.js.gz
[SPIFFS] upload : C:\Users\sebastian\AppData\Local\Temp\arduino_build_16777/esp-rfid.spiffs.bin
[SPIFFS] address: 0x100000
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : COM5
[SPIFFS] speed  : 115200
error: Failed to open COM5
error: espcomm_open failed
error: espcomm_upload_mem failed
SPIFFS Upload failed!
```

Tried re-plugging from USB already.
Any idea how to manually create the auth & fonts folder?
Or can I just flash the latestspiffs.bin on top of the latest.bin via esptool.exe with the same command?

`esptool.exe -vv -cd nodemcu -cb 921600 -cp COM5 -ca 0x00000 -cf latestspiffs.bin`

omersiar, 2017-09-03:
They are actually not folders but filenames itself. Choose file to upload, before hitting the upload button, change filename to /auth/xxxxx.xxx or /fonts/xxx.xxx

MrMontesa, 2017-09-04:
Thanks for the hint. This got me going so I was able to finish the SPIFFS upload. After configuring the device and rebooting, the gui looks fine.
Unfortunately the Users & Logs üage keep loading for ever. Seems like they are not able to retrieve any data.
Also swiping with an RFID tag is not recognized. Even though I'm quite sure cabling and stuff is properly done.

omersiar, 2017-09-04:
which browser are you using? browser's console may have information about the issue which you can open with F12 key on keyboard if your browser is supporting it.

MrMontesa, 2017-09-04:
I'm using Chrome but can flip to Firefox as well.
Console is trying to get info from ```ws://192.168.23.136/ws``` which is in state "pending". Seems like this request is in an infinite loop or time out as it never returns any value and stays there forever.

[image]

omersiar, 2017-09-04:
this ws behaviour on chrome is normal. can you manually add an user? you should see frames as browser sending commands to esp. i tested every browser i can test, so far it worked both on mobile and desktop. i believe your problem is not websocket or browser. fresh install may cause this problem since you already got into problems with spiffs already you may try to flash it with different approach.

MrMontesa, 2017-09-04:
I cant add a users manually as I cant find the "New User" button. 
[image]
Behavior is the same in Chrome/IE/FF.

Then I'll try to flash from scratch. Would you mind sharing a hint on the process of flashing latestspiffs.bin?
First I have to flash latest.bin using esptool.exe, right? How do I proceed with latestspiffs.bin? 
Also flashing with esptool.exe? Do I need a different command?

Alternativly I would be more than happy using your most preferred way of flashing the Wemos D1 mini.
Thanks much for all your help on my crazy problems. Appreciate your help!

omersiar, 2017-09-04:
I have updated files use "ESP-RFID Flash.bat" if you are on a Windows PC. You are clearly having this issue because of difference between firmware and web files.

https://github.com/omersiar/esp-rfid/tree/master/compiledbin

MrMontesa, 2017-09-05:
This worked like charm. Thanks a lot for your help and patience.
After flashing both bins the right way, my device started working just fine.
Closing here.

closed 2017-09-05 by MrMontesa