omersiar/esp-rfid #16: Support for MQTT?

MrMontesa, 2017-09-05:
Since now my setup is working flawlessly now, I do have the first feature request :-)
Could you imagine making MQTT message sending optionally available? 
Here's my idea:
 I would like to send the UID of a scanned RFID tag via MQTT to my broker which is a home-automation system. There I'd like to do some more actions, like turning on lights, starting to play music, etc.

This is is how I could imagine an implementation:
- enhance the "setting page" with an additional form. The following fields are needed: 
  - **MQTT host** containing an IP address. 
  - **MQTT topic** which is a string like "my/special/topic"
  - **MQTT user** optionally
  - **MQTT password** optionally in case one is using a public MQTT server on the internet
- As soon as the MQTT host variable is filled, the MQTT sending should be activated.
- As soon as an RFID token is successfully scanned, the MQTT message should be sent to the configured broker using the configured topic
- The most commonly used MQTT library is: [arduino-mqtt](https://github.com/256dpi/arduino-mqtt) and from the examples it seems to be quite easy to implement

This would really boost functionality.
Let me kwow what you think.
Thanks much

omersiar, 2017-09-06:
https://www.bountysource.com/issues/49012783-support-for-mqtt

#5 #12 #2

omersiar, 2017-09-06:
Give it a bounty, and hope for the best. I would prefer it to be in an another branch.

olskar, 2017-09-22:
Would love this as well

omersiar, 2017-09-27:
For future reference https://bitbucket.org/xoseperez/espurna

electrontube, 2017-09-29:
I would also love to see MQTT Support.

@omersiar 
Many thanks for the great project!

goermezer, 2017-11-18:
Me too. I need it for Smart Home (OpenHAB in my case). A predefined URL call to a HTTP service with the UID as an url parameter could be a workaround if MQTT is too complicated.

rneurink, 2017-11-19:
MQTT is not that difficult to implement the problem here is security. MQTT supports tls but the esp is too weak to perform these tasks so all data will be sent unencrypted. This of course may not be a problem if you use additional security features with reading the cards like a block with a specific key. But in this state if someone intercepts de message you can easily write it to a card and get in. (and yes i am aware that you can also very easily copy cards) basicly if you have a bit of programming knoweledge you can implement this with a few lines of code as it shouldn't interfere with most of the functions

thunderace, 2017-12-03:
Done. See #26

omersiar, 2017-12-04:
See #26

closed 2017-12-04 by omersiar

9acca9, 2018-06-05:
Hi. Can somebody give me a hand to config mosquitto?? i already installed on a Ubuntu 16.04
Thanks