omersiar/esp-rfid #17: Hi

ivanlunardi, 2017-09-07:
I have a problem with websocket on OSX Sierra!
" the message sent directly in websock.onopen doesent work , if you try by console it work"

Ivan

omersiar, 2017-09-07:
Hi,

Can you share Safari version? And are you building esp-rfid from source or flashing compiled binaries?

ivanlunardi, 2017-09-07:
Hi,
Safari Version 10.1.2 (12603.3.8)
Chrome Version 60.0.3112.113
I tried from sourso and with binaries.
I Tried also with other Mac with same issues.

Ivan

omersiar, 2017-09-07:
Do you get same behaviour on chrome too?

ivanlunardi, 2017-09-07:
Yes I Do, the issue is present in every page. 
Ivan

omersiar, 2017-09-07:
goto HTML editor "http://ipaddressofESP-RFID/edit" then select "log.js" delete everything paste below code and hit CTRL+S to save the file and goto "Logs" section.

```javascript
var websock;
var timezone;
var devicetime;
var userdata = {};

function twoDigits(value) {
   if(value < 10) {
    return "0" + value;
   }
   return value;
}

function initTable() {
  jQuery(function($) {
    FooTable.init("#latestlogtable", {
        columns: [
            {
            "name": "timestamp",
            "title": "Date",
            "parser": function(value) {
              var vuepoch = new Date(value * 1000);
              var formatted = vuepoch.getUTCFullYear()
                  + "-" + twoDigits(vuepoch.getUTCMonth() + 1) 
                  + "-" + twoDigits(vuepoch.getUTCDate())
                  + "-" + twoDigits(vuepoch.getUTCHours())
                  + ":" + twoDigits(vuepoch.getUTCMinutes())
                  + ":" + twoDigits(vuepoch.getUTCSeconds());
                return formatted;
            },
            "sorted": true,
            "direction": "DESC"
            },
            {
            "name": "uid",
            "title": "UID",
            "type": "text",
          },
          {
            "name": "username",
            "title": "User Name or Label"
          }
        ],
        rows: userdata
      });
  });
}

function start() {
  websock = new WebSocket("ws://" + window.location.hostname + "/ws");
  websock.onopen = function() {
	alert("WebSocket Opened");
    var commandtosend = {};
    commandtosend.command = "latestlog";
	alert("Now i will try to send latestlog command");
    websock.send(JSON.stringify(commandtosend));
    commandtosend = {};
    commandtosend.command = "gettime";
	alert("Now i will try to send gettime command");
    websock.send(JSON.stringify(commandtosend));
  };
  websock.onclose = function(evt) {};
  websock.onerror = function(evt) {
    console.log(evt);
  };
  websock.onmessage = function(evt) {
    var obj = JSON.parse(evt.data);
    if (obj.type === "latestlog") {
      userdata = obj.list;
      initTable();
      document.getElementById("loading-img").style.display = "none";
    } else if (obj.command === "gettime") {
      timezone = obj.timezone;
      devicetime = obj.epoch;
    }
  };
}
```

ivanlunardi, 2017-09-07:
I find the solution,
the problem is the variable socket you mast set it to null in declaration
var websock; -> var websock = null;

I tried with a demo of node application and i had the same problem.

Ivan

omersiar, 2017-09-07:
Thank you finding this bug. Kind regards.

closed 2017-09-07 by omersiar