omersiar/esp-rfid #18: Can't build it...

nardev, 2017-09-08:
I tried to build it, installed all required libs but i got this error.

```
src/main.cpp:317:14: error: 'class ESP8266WiFiClass' has no member named 'scanNetworksAsync'
WiFi.scanNetworksAsync(printScanResult, true);
```

Any idea what could be an issue?

omersiar, 2017-09-08:
You need git version of esp8266 arduino core

rneurink, 2017-09-08:
You need to have the latest esp arduino package and you don't have that one

closed 2017-09-08 by nardev

ersinpw, 2017-09-09:
Where did you place the git version of the esp8266 Arduino core?

Mine is located here (osx): 
` /Users/ersin/Library/Arduino15/packages/arduino/hardware/esp8266com/esp8266/`
I guess it is the correct path according to the manual:
https://github.com/esp8266/Arduino#using-git-version
but fortunately I still see the same error