omersiar/esp-rfid #19: Question?

nardev, 2017-09-08:
Did you test the log size and efficiency with a lot of users/cards in the file? How did it behave?

p.s. I plan to make wiegand version and slightly different code structure, is there a way to contact you other than this messages?

nardev, 2017-09-08:
p.p.s it's sometimes slow and unstable when opening web interface, i know by other ESP projects that it happens sometimes.... but is there anything which can improve the web stability?

nardev, 2017-09-09:
:D interesting, it opens web interface quite fluently from windows machine...

and.. now i'm quite confused... 

What could me an issue???

omersiar, 2017-09-09:
Thank you for your feedback. I did not tested logging yet but it is on my to do list. we have a Gitter chat room, see readme for the link.

if you try to connect to web ui via mdns hostname,  there is a delay, your pc tries to resolve hostname, and it does that second time to open websocket connection. try to connect via ip, see known issues.

i think web ui is stable, but it definetly can be improved. pages will be AJAX loaded, websocket connection will be always opened etc. they are also on my todo list.

i need feedbacks, just use it and let us hear your findings.

closed 2017-09-09 by nardev

led707, 2025-03-28:
Hi all. 
I have issue about recognize the codes 
At the users, adding new chip, it got idebtifyed with 'fffed' name. 
Not with the '12345' (5 digits on the chip side).
This problem cant help me to block lost chips, beacuse cant identify it.
Can anyone help mehow to di it? 
Thanks in advance.