omersiar/esp-rfid #20: ESP-IO

Pako2, 2017-09-11:
Hi Omer!
I want to thank you for this project. I'm a newcomer to ESP8266 and I was looking for how I should start. Your project was perfectly suited to my needs and I used almost everything.
You can see the result here: [ESP-IO](https://github.com/Pako2/EventGhostPlugins/tree/master/ESP-IO)

Note:
I have solved (at least for my needs) two security holes:
1) Entering the editor is password protected. I do not suppose that it is necessary to open the editor if the device works in "Fallback to AP Mode".
2) (Direct) access via the websocket is also password protected. This is more or less a "scam". If a JavaScript is triggered (and this happens only if the user enters the admin password), it first opens EventSource. The server will reply to this opening by sending a password. Websocket communications can then be started by asking for a password (from the server side). So if I want to open directly websocket (for example, using Eventghost) then I need to know the password. Otherwise, the server will disconnect me.
However, I'm aware that this is just a minimal protection and it can not be used where real system attacks can be.

Thanks again for your project!

Pako

akeilox, 2017-09-11:
Hi Pako

Cant help but being curious on what kind of application do you use it with Eventghost? I recall messing with eventghost on vista and mce-remote days and havent touch since then...
I noticed there is also RDM6300 rfid reader and lcd display plugins for Eventghost, along with ESP-IO in your repository. And wondered how one can use these with eventghost, what would be the applications like.

Can you share on the type of application(s) do you use these, in coordination with Eventghost?

Pako2, 2017-09-11:
Unfortunately, I do not understand the question.
In what context should I use some collaborative applications with Eventghost?
I can not answer such a question. This is too general.
For example, I can say that I use the MPC-HC player, which is controlled using the remote controller just via EventGhost. To do this I need two plugins. One for remote control and one for MPC-HC. When I press a button on the remote control, it will trigger an event in Eventghost. This event is assigned to a macro where some action of MPC-HC plugin is located ...

I maybe will be able to answer better, when you ask for a specific problem.

Pako

akeilox, 2017-09-11:
Hi Pako
What I meant to ask is what kind of use case do you have with the RFID reader RDM6300 as EventGhost plugin.

I know, briefly, EventGhost used as home automation, mainly with remote control, but I was wondering what kind of things (real life application) one can do using RFID reader with EventGhost. I am simply curious and trying to understand the type of applications possible with it.

Pako2, 2017-09-12:
So you mean specifically RDM6300 with EventGhost. So it's really quite simple. Nothing else is needed except for a relay board (for which exist an EventGhost plugin). This can be, for example, [ICS Relay](http://www.eventghost.net/forum/viewtopic.php?f=9&t=6955) or [HID relay](http://www.eventghost.net/forum/viewtopic.php?f=9&t=7426).
ESP-IO could also be used (a relay has to be connected, switched by a transistor).
As a control program, Eventghost is completely self-sufficient, nothing more is needed.

Pako

omersiar, 2017-09-12:
At some point i realized that ESP-RFID would be a good starting point for HTML WebSocket applications. HTML Editor is not meant to be stay as a feature. It will be dropped eventually, it gives ability to rapid development for tinkerers like you. I am trying to figure out securing the whole project, your approach similar to authentication challenge which is i believe supported by the original ESPAsyncWebServer library.

Nicely done with ESP-IO.

https://jjssoftware.github.io/id8266-aka-skynet/

nardev, 2017-09-15:
@Pako2 nice project :)

closed 2017-12-24 by omersiar