omersiar/esp-rfid #21: Can't build esp-rfid

ersinpw, 2017-09-20:
Similar to issue #18 

Where did you place the git version of the esp8266 Arduino core?

Mine is located here (osx):
/Users/ersin/Library/Arduino15/packages/arduino/hardware/esp8266com/esp8266/
I guess it is the correct path according to the manual:
https://github.com/esp8266/Arduino#using-git-version
but unfortunately I still see the same error

omersiar, 2017-09-27:
I do not use mac, maybe @rneurink may help on this.

rneurink, 2017-09-28:
One of my testers has a mac but i haven't been building on it. If you go
into the arduino settings and put the url in additional board manager. You
are able to select the right version for the arduino core in the library
manager

jugs695369, 2017-10-08:
Plz provide working video or detailed operation of process

closed 2017-10-11 by omersiar