omersiar/esp-rfid #22: security issue of websocket

simoncheen, 2017-10-20:
hello, 
we found a security issue of the websocket.
```step 1. ```
 http://www.websocket.org/echo.html
fill 
```bash
ws://ip_address/ws 
```
and click connect.

```step 2.```
fill 
```bash
{"command":"getconf"}
``` 
in the message text box , and click "send" button. 

wow, you would get the password

omersiar, 2017-10-20:
Also you can delete a known Users, change configuration, etc. Simply WebSocket communication is not secured at all. See [ToDo](https://github.com/omersiar/esp-rfid#to-do) list. I should definitely mention this at Known Issues level. Thank you.

simoncheen, 2017-10-20:
I'm try to change the websocket as this https://github.com/Links2004/arduinoWebSockets , it seems have basic Authorization

omersiar, 2017-10-28:
See https://github.com/me-no-dev/ESPAsyncWebServer/pull/272

simoncheen, 2017-10-28:
perfact !

commit ab549d07 (2017-10-28): This should fix #22

This also need to be fixed on library level.

Update ESPAsyncWebServer library and apply the patch.

See https://github.com/me-no-dev/ESPAsyncWebServer/pull/272
closed 2017-10-28 by simoncheen