omersiar/esp-rfid #24: My mfrc522 does not seem to read the card

1330154737, 2017-11-13:
Hello, I am a novice learning esp8266, I follow the steps to connect the hardware, in the user interface in accordance with the format of the picture inside the card input UID, set the web page test button can be triggered, but the MFRC522 reader part did not respond
`[ INFO ] ESP RFID v0.3alpha
[ INFO ] Config file found
{
  "command": "configfile",
  "bssid": "74:7D:24:36:C3:C2",
  "ssid": "junhao",
  "wmode": "0",
  "pswd": "10086520",
  "sspin": "15",
  "rfidgain": "96",
  "rtype": "0",
  "rpin": "4",
  "rtime": "300",
  "ntpserver": "pool.ntp.org",
  "ntpinterval": "30",
  "timezone": "8",
  "hostnm": "admin",
  "adminpwd": "admin"
}
[ INFO ] Trying to setup RFID Hardware
[ INFO ] RFID SS_PIN: 15 and Gain Factor: 96
[ INFO ] MFRC522 Version: 0x92 = v2.0
[ INFO ] Trying to connect WiFi: junhao....
[ INFO ] Client IP address: 192.168.123.150
`

omersiar, 2017-11-13:
Hello,

Did you try Dumpinfo example from MFRC522 library?

1330154737, 2017-11-14:
I tried burning mfrc522 example program, it did not work, but I found a problem that is even after burning mfrc522 test program can be seen in the router interface esp8266 connected to the wifi @omersiar

omersiar, 2017-11-14:
[ INFO ] MFRC522 Version: 0x92 = v2.0

This means you succesfully connect the RFıD reader, it may be your rfid tags (piccs) are not for the reader. Try find mifare classic tag

1330154737, 2017-11-14:
Thank you for your reply, I use dumpinfo examples can read the card, but I follow the example connected rst pin

1330154737, 2017-11-16:
Thanks for your project, I modified Antenna Gain and now it's working

omersiar, 2017-11-16:
Hello, great! Which gain value did you chose?

1330154737, 2017-11-16:
I chose 43😃

closed 2017-11-17 by omersiar