omersiar/esp-rfid #30: Question Wiegand Reader

SamNitro, 2017-12-10:
Hey,
Is it possible to include the wiegand readers?

omersiar, 2017-12-11:
It is indeed. Maybe @nardev has some taughts about it.

nardev, 2017-12-13:
If @omersiar wants to merge it, i'll make it as new commit, but in that case, @omersiar should change the project description and plan?

@omersiar would you do that?

At the moment, it's not 100% stable, just changed the current code to work with wiegand readers and that works great.

omersiar, 2017-12-14:
Project itself getting more complex with each feature that is supported, each feature should can be enabled or disabled at compile time, but that needs more robust build system. I have a plan to change WebUI structure to support plugins like MQTT, Wiegand, etc.

@nardev I can merge it on another branch, so people can test it more widely, so we can fix bugs if there is any then it can be merged to master branch. I will make changes at description and change log.

nardev, 2017-12-14:
@omersiar yea.. that is good.

Regarding  UI.. i believe it can be much more  simple...

@omersiar have you thought about moving to platformio structure?

SamNitro, 2017-12-16:
That looks like good, thanks for the great work.

amp83, 2017-12-17:
2 questions is there a way to get esp to load settings if MFRC522 is not connected. (Haven’t arrived yet)
2nd any way of using ibuttons (onewire devices) instead of MFRC522

teriall, 2017-12-20:
Hello! First of all thank you for the great project! I tried to use it as a door lock and I have a couple of questions: Is it possible to add a buzzer to make a sound when access allowed or denied and a button to unlock the door from inside for example and from settings page maybe? Is it possible to realize it for donation? :)

nardev, 2018-01-12:
@omersiar i did some code,  just tell me what is your final intention:

1. Only wiegand based branch
2. Make it optional in settings, to choose a reader and reader pins

I can make both things.. currently i have installed few wiegand based readers, works fine but i use interrupts

Is there any particular reason not to use interrupts for you?

omersiar, 2018-01-12:
Let's do 2, because the other branches won't get any updates unless someone manages it. The reason we did not use interrupt on MFRC522 is because of complexity. I never managed it working.

https://github.com/miguelbalboa/rfid/blob/master/examples/MinimalInterrupt/MinimalInterrupt.ino
https://github.com/miguelbalboa/rfid/issues/311

akeilox, 2018-01-13:
its great to hear Wiegand support in the horizon, there is so much potential in converting old oem readers into modern  mqtt

SamNitro, 2018-01-15:
@nardev I have read you are testing the Wiegand Reader... if you have a compiled bin I can test it too :)

nardev, 2018-01-15:
@SamNitro you can test it now: https://github.com/nardev/esp-rfid

To me it worked for both readers.

SamNitro, 2018-01-16:
Hey, I can't save the Wiegand settings. After Reboot it is unchecked.

nardev, 2018-01-16:
@SamNitro i'll test it,  just a sec...

nardev, 2018-01-16:
@SamNitro oh, it saves the settings it was just when i did clean up.. i checked if the  readerType is 0 istead of "0"... before i used == and there was a warning in commit.. 

feel free to continue... it actually save it.. i'll pus the change now...

nardev, 2018-01-16:
@SamNitro please tell me  if it works as a reader.. is everything fine?

closed 2018-01-16 by nardev

SamNitro, 2018-01-16:
No I'm sorry, he lose the GPIO information. He set D-0 and D-1 back to GPIO-5
I will use 
D-0 -> GPIO 14
D-1 -> GPIO 12
on Demos D1 mini

nardev, 2018-01-16:
ah.. it's just predefined in form... sorry... i'll fix that too

once you choose it and set it.. just "save/reboot" and it should work.. despite what it shows in settings.

nardev, 2018-01-16:
@SamNitro thank you for testing... it works now...i just pushed.. there was slight mistake in HTML form

SamNitro, 2018-01-16:
Perfect, it works :)
Thx for support and work.

nardev, 2018-01-16:
@SamNitro glad to hear that....

@omersiar sorry.. as the functions pile up... more testing is required... :) probably once we simplify the frontent, it will be less mistakes of this kind...

JuculianD, 2018-02-28:
I would like the wiegand reader function as well. I got an 125khz wiegand reader pcb with antenna from aliexpress and I would like to use it. However, the prebuilt builds do all not contain the wiegand settings on the settings page. I will wait for the new frontend.