omersiar/esp-rfid #31: Sync data with a remote MySQL database

prassaschr, 2017-12-10:
Hi.
Please include syncronization with a remote MySQL.
This way the log file, which is JSON, might also stored to a MySQL database for security purposes and/or backup.
This way the system can be used as an attendance logger.
Thanks

omersiar, 2017-12-11:
Hi @prassaschr ,

ESP-RFID self sufficient project. Implementing MySQL support is easy but this will require specific solution for each MySQL setup, instead you can query ESP-RFID with a Node.js application (using WebSocket client) or any other tool for record keeping. Node.js and MySQL + WebSocket packages is just enough for the task.

Kind regards.

closed 2017-12-11 by omersiar

prassaschr, 2017-12-11:
This sounds interesting! Could you guide me by posting an example URL? I am not familiar with Node.js and Websocket client.
Thank you once more.
Kind Regards

omersiar, 2017-12-11:
For example this is a simple Node.js script that is Websocket Server and a MySQL client. Websocket server listens for queries and executes a search query on MySQL server, then sends back results to Websocket client.

```javascript
var mysql = require('mysql');
var connection = mysql.createConnection({
  host: 'localhost',
  user: 'admin',
  password: '1',
  database: 'tek_servis'
});

const WebSocket = require('ws');
const wss = new WebSocket.Server({
  port: 8080
});

wss.broadcast = function broadcast(data) {
  wss.clients.forEach(function each(client) {
    if (client.readyState === WebSocket.OPEN) {
      client.send(data);
    }
  });
};

function sorgula(data) {
  connection.query("select c.hesapkodu, c.cariadi, c.tipi, c.cinsi, c.marka, c.model, c.tamadi, c.sfiyat, c.afiyat, c.ptipi, c.barkod, (case when 1=1 then(SELECT sum(coalesce(d.miktar,0)*m.gc) FROM stokgcm m, stokgcd d WHERE m.gcsno=d.gcsno and m.aktif=1 and d.hesapkodu=c.hesapkodu) else 0 end) as Kalan from cariler c where c.hesapkodu like '153%' and c.bayino=1 and c.aktif=1 and (c.barkod='" + data + "') or (c.tamadi like '%" + data + "%')", function(error, results, fields) {
    if (error) throw error;
    wss.broadcast(JSON.stringify(results));
  });
}

wss.on('connection', function connection(ws) {
  ws.on('message', function incoming(message) {
    var obj = JSON.parse(message);
    if (obj.type === "sorgu") {
      sorgula(obj.data);
    }
  });
});

```