omersiar/esp-rfid #35: Doesn`t detect NFC of my Smartphones

goermezer, 2017-12-19:
Hi,

thank you for this cool project. Fits exactly to my needs. It's working successfully with my NFC Badge, Keyring and Stickers. But unfortunately it doesn`t work anymore with my Smartphones (Samsung S6 and Huawei P10 Lite). Whenever creating the user by holding the smartphone to the reader, it always generates a new ID. Each try seems to generate a unique ID. 

When testing with the first compiled binary both smartphones worked. But after the last version (or before) it doesn`t work anymore.

omersiar, 2017-12-19:
Hello,

Are you emulating the RFID tag on your phone? This may be caused by RFID library changes.

Jorgen-VikingGod, 2017-12-19:
As I now, the cheap MFRC522 RFID module is not able to read active tags (=NFC from Smartphones)
You have to use the PN532 RFID module (it costs about 4 USD, but requires a different library to access it over SPI/I2C/UART)

If you need a cheap setting, the RC522 is great. But for stronger signals and longer read range, I really recommend the PN532.

goermezer, 2017-12-19:
I`m not an expert regarding RFID. I heard NFC is similar to RFID. So I simply tried it with one of the first versions and it luckily worked. But if MFRC522 is not designed for that kind of signals, I`ll be still happy with keyrings, stickers and what ever kind of tags...

Thanks for the info.

closed 2017-12-24 by omersiar