omersiar/esp-rfid #38: Can't get this working correctly... Hopefully you can help!

bignellg, 2017-12-27:
Hi,

I'm a noob and trying to get this working. Can you confirm if the following board should work? https://www.amazon.co.uk/dp/B01N5D3MV8/ref=pe_3187911_185740111_TE_item

I am able to flash the board and log into the web interface but I see this in the serial monitor:

[ INFO ] Trying to setup RFID Hardware
[ INFO ] RFID SS_PIN: 15 and Gain Factor: 32
[ INFO ] MFRC522 Version: 0x0 (unknown)
[ WARN ] Communication failure, check if MFRC522 properly connected
[ INFO ] ESP-RFID is running in AP Mode 
[ INFO ] Configuring access point... Failed!
[ INFO ] AP IP address: 192.168.4.1
[ INFO ] AP SSID: test
[ INFO ] ESP-RFID is running in Fallback AP Mode
[ INFO ] Configuring access point... Ready
[ INFO ] AP IP address: 192.168.4.1
[ INFO ] AP SSID: ESP-RFID-1cc374

I have used this RC522 on another (more simple) project using an uno and it works fine (can scan cards and work a relay), so i'm not sure if what the issue can be as it's even failing to setup the AP mode.

bignellg, 2017-12-27:
Also if I use client mode I am able to connect it to my network but for some reason NTP isn't working and I have the same issue with the RC522 card reader.

omersiar, 2017-12-28:
It seems there is a bug with the AP mode. Your board should work, there is no doubt there.

bignellg, 2017-12-28:
Hi,

I have confirmed the RC522 is working by using the dumpinfo and RFID_Access sketches successfully. I have doubled checked the connections but still unable to get any success with your sketch. I have also tried changing the SS Pin to another option (same as used in the dumpinfo)... Any thing I might be missing?

omersiar, 2017-12-28:
Test Gain Factor, try other options.

bignellg, 2017-12-28:
Unfortunately trying different Gain Factors didn't work. I have ordered another RC522 just in-case it is faulty and will hopefully have it tomorrow to continue testing.

bignellg, 2017-12-29:
The new RC522 works :)... Trouble now getting the relay to work :(....
Can you share a link to a relay that should work (just in case I have the wrong kind)...

omersiar, 2018-01-06:
http://www.instructables.com/id/Controlling-AC-light-using-Arduino-with-relay-modu/

closed 2018-01-06 by omersiar