omersiar/esp-rfid #39: Can not retreive logs anymore

goermezer, 2017-12-28:
Since a couple of days I can not retreive the logs anymore. The page simply says "please wait while fetching data". But nothing happens.

It seems, that there is no possibility to do a hard reset. I reflashed the firmware remotely with the upload feature or did power cycles. But no way...

The problem is, that I have put the whole stuff into a housing and mounted it to the garage door. To do a factory reset or a similar feature would be a great help...

Thanks
Mustafa

omersiar, 2017-12-28:
This may caused by SPIFFS file system also need to be flashed. But there is no easy way to do this without computer connection.

goermezer, 2017-12-28:
Oh, that's hard :-( But I see that factory reset is already on the todo list. From my point of view it`s one of the most important features to prevent users from unmounting the whole stuff only to flash it once again. 

But I have one more issue with MQTT, which possibly have a hickup: first it worked for me reliably - until I renamed the topic in the settings. After that it seems not to publish the topic to mosquitto anymore. I checked it by subscribing the configured topic with MQTT.fx from Windows (where also mosquitto runs). But no messages arrive. Also the topic collector feature could not catch it from the ESP. And subscribing to # doesn`t list my configured topic.  Any ideas regarding this? Or do I really have to carry my laptop to the garage - for the amusement of my neighbours ?

omersiar, 2018-01-06:
I simply amazed by you already deployed esp-rfid to your garage. There is no check for web files if they are also updated.

It's nice to have a real world usage case. I can not tell much about MQTT, maybe @thunderace can help with this.

OTA SPIFFS update would be nice but not every esp is connected to Internet, also we definitely need some error handling with the web files compatibility.

- [ ] Check SPIFFS version after update then warn admin at Web UI and also on terminal if there is a mismatch or no backward compatible change is made.

thunderace, 2018-01-06:
> And subscribing to # doesn`t list my configured topic.

It's because the topic used by esp-rfid is not 'retained' so you can see it only when someone publish something to it.

> The page simply says "please wait while fetching data". But nothing happens.

It's related to frequent websocket close that i see every x minutes on my system. I'll publish soon a fix for that (but i can't find the source of the problem...)

Could you use the embedded editor to see the file content and check that the configuration file content is OK?

goermezer, 2018-01-06:
@omersiar : 
> I simply amazed by you already deployed esp-rfid to your garage. There is no check for web files if they are also updated.
Happy to amuse you :-) But it looked quite stable on my desk. So I deployed it after some days. In fact, the rfid mechanism works very well so far. I never stood in front of a closed garage door which isn`t opening - even though my wife expected something different...

goermezer, 2018-01-06:
@thunderace 

> you can see it only when someone publish something to it.

ok, I will look at this later - when I have fixed my weak WIFI signal at the garage door. At the moment I haven`t seen any message from ESP-RFID. Neither for my configured topic nor at # when I listen for with MQTT.fx in the moment I open the garage door.

 did this only fBut whenever the garage door is opened, the ID of the transponder should be sent. This worked for a couple of days before.

> It's related to frequent websocket close that i see every x minutes on my system. I'll publish soon a fix for that (but i can't find the source of the problem...)

I have this behavior all the time. No single line of the table can be loaded.

> Could you use the embedded editor to see the file content and check that the configuration file content is OK?

That`s new to me. I uploaded the compiled binaries and I can not see an editor implemented anywhere. Does only the GIT checkout contain it?

thunderace, 2018-01-07:
You can, acces to the embedded editor with this url : http://yourespip/edit user:admin password:admin
With the editor you can check see/edit all files on the esp.
Check the configuration file : /auth/config.json

goermezer, 2018-01-07:
That`s a good help to access the files. But the password need to be changed. 

My config looks good. It`s the same content like downloaded settings.json. But my latestlog.json is empty. And the files below /P/* are not accessible. When I select them, the editor doesn`t switch to the content of these files.

goermezer, 2018-01-07:
With the editor I can delete and create files. So I deleted latestlog.json and accessed the garage door. After that the latestlog.json was automatically generated with this content:

{"type":"latestlog","list":[]}

After the next accesses, the latestlog.json gets filled with log entries and the page works. So this problem is solved :-)

goermezer, 2018-01-09:
One more feedback from a real world use case:

I could also solve my last issue with MQTT, which I mentioned in my second post (messages were not published). For this I simply reflashed the ESP two times: first with the latest.bin of omersiar (without MQTT) and then once again with the latest.bin of thunderace (incl. MQTT). 

However, MQTT works again. I`m perfectly happy now.

omersiar, 2018-01-18:
@goermezer Mustafa, thank you for your feedbacks. This behavior will be changed on upcoming releases.

closed 2018-01-18 by omersiar