omersiar/esp-rfid #46: Adding buzzer

nedoskiv, 2018-01-12:
Hello,
ESP-RFID will be better if we add a buzzer attached to an GPIO and send short sound signals, on reading, giving access/deny access , powerup, ЕТC. I can offer my help for that. I got some (above) basic arduino programing experience, but i'm good at algorithm creation.

teriall, 2018-01-13:
Hi! Will be nice if you add that kind of function. I already ask the about it, but creator of the project busy now.

nardev, 2018-01-14:
By using this board, you will have that option for buzzer/bell. Read the description https://www.tindie.com/products/nardev/esp-rfid-board-for-wiegand-readers-at-5v-to-28v/

I already added wiegand support, just  waiting @omersiar to fix something and merge  it and very very soonthere will be new hardware supporting  the relay opening by my custom board and esp-rfid

closed 2018-01-18 by omersiar