omersiar/esp-rfid #47: Last commit broke the user page...

nardev, 2018-01-14:
Unfortunately, last commits broke the user page. It just stops at a message:

"You can add new users by either scanning a RFID Tag or by clicking on "New User" button."

And there is no add button to be shown etc..

please check, i have Wiegand version waiting for that. It's tested, works fine.. but this is confusing :(

hagelric, 2018-01-14:
Same issue here. "fetching" data freezes...

omersiar, 2018-01-15:
16c6869 should fix this, did not test myself, I hope you guys may test and share the result.

hagelric, 2018-01-15:
No. 
I found the error:
After
`  wsUri =protocol+ window.location.hostname + "/ws"; 
  websock = new WebSocket(wsUri);`

File user users.js needs: (there is no "listener"). Please check log.js too.
`  websock.addEventListener('message', socketMessageListener);
  websock.addEventListener('error', socketErrorListener);
  websock.addEventListener('close', socketCloseListener);`

nardev, 2018-01-15:
No :( didn't help...

Here is what i get in console... https://preview.ibb.co/fckbRm/Screenshot_from_2018_01_15_13_04_44.png" alt="Screenshot_from_2018_01_15_13_04_44

nardev, 2018-01-15:
Did you fix it @hagelric ? Push Commit ???

i can't  dig in.. if  you fixed it, could you just post it ... i want to test the wiegand reader

omersiar, 2018-01-15:
Latest commit should fix it for now. Tested with my setup.

commit 16c68692 (2018-01-15): Fix #47

Fix #47
commit 306cc072 (2018-01-15): Fix #47 for real

Fix #47 for real.
closed 2018-01-15 by omersiar