omersiar/esp-rfid #53: Embedded web files

thunderace, 2018-01-17:
Under the embedded-html branch you have the new esp-rfid version without FS (except for data storage of course).

With the big help from the ESP8266_WiFi_v2.x project. Thanks to them.

TODO : 

- [ ] Authentification
- [ ] One htm file and one js file (it's a work in progress,  i need 2 days of free time to complete)

nardev, 2018-01-17:
@thunderace have you noticed the main.cpp in development branch?

thunderace, 2018-01-18:
Yes I saw but I had already done a lot of work, too late.
And there are several things that make me feel bad in your branch:

- The 'build_flags = -Wl,-Tesp8266.flash.4m.ld' in platformio, it's not the correct one (no spiffs space),
- I do not really see the value of completely changing the contents of the .ino file only to switch from .ino to .cpp. Because of this, when working on another branch based on the old version, it is very difficult to merge its changes ...
- Last point, in my opinion, it is necessary to preserve a branch development which serves to merge the various other branches before the final merge towards the stable branch which must remain the default branch of the deposit.

nardev, 2018-01-18:
Dear @thunderace i was trying to be very careful in making any decision.

This is not my project, i have time and will to contribute but it's really hard if there is no collaboration but stubborn selfish decision making.

1. Just remove those build_flags.. i don't  have any personal issues with it
2. I asked few times about moving to .cpp and why didn't you openly said anything against?
3. If i change some line in readme or if i do comments cleanup.. i see no reason to go through staging. This is very small project.
After all, this is development branch,  i was only pushing  there, when we had master, i didn't want to push anything there until i push first to my fork at least...
4. If you want to be so pushy and not for collaboration and exchange, don't worry, take over, do as you wish, treat this as personal repo .. and contrary to collaboration projects, do as you like, don't consult anyone, pick libs and approaches as you like.

I don't see that it's the way for me.

omersiar, 2018-01-18:
Please just relax, there is a misunderstanding, it's all fine. Let the heat get some cool, i can not even copied MQTT code from its branch, it's just fine, we do not need to rush.

nardev, 2018-01-18:
Kusura bakma...

omersiar, 2018-01-18:
@nardev Önemli değil :)

@thunderace I'm not even a developer, I could not even cherry-pick a commit once, all these git stuff is too complex for me, at least for now though. But i'm a quick learner, please share your best practices, I try to follow.

omersiar, 2018-01-18:
Let's track this #57

thunderace, 2018-01-19:
@omersiar this branch is very different from the all_in_one_html because embedded html is a little tricky and i'am not sure that it's the right direction for the future. It's a proof of concept for now

omersiar, 2018-01-19:
Oh now i see, my apologies.

omersiar, 2018-01-20:
> Gzip required.* files -> FAILED

I may have a guessing why this is failing, browser and the web server need to hand shake over a gzipped content.

I need to look how hex encoded header files are being sent.

omersiar, 2018-02-18:
@thunderace I may find a solution to embed web files into flash. See 3d99110

Everything can be gzipped and be embedded into flash.
There is no need to flash SPIFFS binary, this also opens a solution to do a factory reset since SPIFFS can be formatted, Travis can build a single firmware for the future releases, etc.
There is no impact on loading speed.
SPIFFS completely can be used for the data storage

I will refactor WebUI completely.

I believe this is the right direction, there are multiple gains to benefit from this.

closed 2018-02-28 by omersiar

Netoperz, 2019-03-30:
@nardev 
Hello, One stupid question, I do not want to start new issue just with question, so i ask here
How should i use tools for editing and compiling the webinterface ?

https://github.com/esprfid/esp-rfid/tree/stable/tools/webfilesbuilder

I need to add some stuff to the UI, change colour, some extra text, and add forms for additional topics 
Also need to add forms for setting some stuff like pincode lenght and key value for reading encrypted sectors of the card with other than standard key....

I have edited the 
https://github.com/esprfid/esp-rfid/blob/stable/src/websrc/esprfid.htm
and
https://github.com/esprfid/esp-rfid/blob/stable/src/websrc/index.html

at the start, and what exactly should i do to get the .h files with my changes to build the code and flashit to see how it looks ?

Lest's say that i want to do that using that .exe binary on windows 10 (not the gulp and nodejs way)
Could you point me to the manual/project where the steps are descibed ? if not could you just post here  what exactly should i do ?

Thank you (i have never used that tool and i was always using html in the code, this way is something new for me)