omersiar/esp-rfid #54: Flash compiledbin error

SamNitro, 2018-01-17:
Hey i try to flash the compiledbin (stable and embedded-html) with windows flash.bat

After connect to 192.168.4.1 it open a site with "Not found"

What am i doing wrong

nardev, 2018-01-17:
@SamNitro it occurred to me few times..

there is no filed in the SPIFS  ... due to some wrong settings in ld scripts it can happen that it will not properly upload code and html/js files.. once you do one, the other is wiped out..

try to comment this line in platformio.ini

build_flags = -Wl,-Tesp8266.flash.4m.ld

SamNitro, 2018-01-17:
How I upload correctly with platformio?
Can I easy click on upload an done?

SamNitro, 2018-01-17:
i have forget to flash 
pio run --target uploadfs

Now it run
Sorry :/

closed 2018-01-17 by SamNitro