omersiar/esp-rfid #57: all in one HTML/JS/CSS files

thunderace, 2018-01-18:
done in the "all_in_one_html" branch.
Need some work/test for backup and restore.

SamNitro, 2018-01-18:
i have another issue
after flash I can not change everything in settings page
he says "please wait while fetching data..." and nothing happens

omersiar, 2018-01-18:
I will test this.

omersiar, 2018-01-18:
`src\web_server_static_files.h:2:36: fatal error: web_server.required.js.h: No such file or directory`

omersiar, 2018-01-18:
After installing Java, now this: `[src\esp-rfid.ino.cpp] TypeError : unsupported operand type(s) for -: 'float' and 'NoneType'`

I did not know the Java is required.

omersiar, 2018-01-18:
I do not know how I built it but now i can flash, but i can not access web pages through my phone.

omersiar, 2018-01-18:
- [ ] Place a warning if there is no Java available during pio_script.py:19  `if java_installed:`

omersiar, 2018-01-18:
Both required.js and .css files are actually coming from JQuery, Bootstrap, FooTables, it will be nice to have them separately and combine them during compile time, so they can be updated whenever they receive an update.

SamNitro, 2018-01-18:
can you edit anything on the settings page?

SamNitro, 2018-01-18:
[image]

thunderace, 2018-01-19:
@omersiar i think you're not on the right branch : java is used in the "Embedded web files" branch, not this one.
And if java is not installed there is a fallback to inline minify so you need an internet connection.

@SamNitro if you cannot edit the settings page it's because the config file is not loaded (for whatever reason).

SamNitro, 2018-01-19:
I don't now, it will not work.

thunderace, 2018-01-19:
@SamNitro My bad, i fix something for the AP mode. I haven't see the bug before because i never use this mode (i always upload my config file).
Try the last commit. It will work now...

SamNitro, 2018-01-19:
Perfect :)

Now I'm just waiting for MQTT and Schedule User Access then I'm happy ;)

thunderace, 2018-01-19:
Hum... another bug fix in the last commit... sorry

SamNitro, 2018-01-20:
if you add a user it's not show in the page tab "Access Type". After reload page it's correct

thunderace, 2018-01-20:
Do you use the last commit?. you know, i make many bug fixes so look frequently the commits.

SamNitro, 2018-01-20:
Yes it‘s the last commit. I check regularly if there is something new ;)

thunderace, 2018-01-20:
Sorry to ask this but do you upload the SPIFFS at each new commit?

thunderace, 2018-01-20:
I just upload the spiffs and firmware from the last commit and i see no error on user add...
Strange

SamNitro, 2018-01-20:
Yes i Upload Both from last commit. 

Add a new User and Label it Activ and than change the the view only Activ User

omersiar, 2018-01-20:
<link rel="icon" href="/favicon.ico">
    <link href="/required.css" rel="stylesheet">
    <script src="/required.js" type="text/javascript"></script>
    <script src="/auth/settings.js" type="text/javascript"></script>
    <script src="/auth/users.js" type="text/javascript"></script>
    <script src="/auth/log.js" type="text/javascript"></script>
    <script src="/auth/esprfid.js" type="text/javascript"></script>

Can your Browsers load all these resource with one shot?

thunderace, 2018-01-20:
> Add a new User and Label it Activ and than change the the view only Activ User

Ok, i see. working on this...

SamNitro, 2018-01-27:
hey thunderace do you have time to add mqtt support?

thunderace, 2018-01-27:
@SamNitro : no i can do nothing for now : my house is flooded and i'am in it (France, La Seine, floods).

SamNitro, 2018-01-27:
oh I'm sorry. then I hope that is not too much broken.

omersiar, 2018-01-27:
@thunderace sorry to hear, i hope everthing get better

closed 2018-02-28 by omersiar