omersiar/esp-rfid #58: Please wait while fetching data...

Beardy241, 2018-01-23:
I've got this joining my wifi and initialising the MFRC522 — so far so good.

I'm stalled now on the web UI with it saying "Please wait while fetching data..." — which it never does.

I've tried re-uploading the SPIFFS a few times.  No joy.

Thoughts?

(Ace project by the way!  Hoping to use it on our new Makerspace and Men's Shed)

omersiar, 2018-01-23:
Did you use arduino code? https://github.com/omersiar/esp-rfid/releases

Beardy241, 2018-01-23:
Yes.

omersiar, 2018-01-23:
There is a problem with the log page. It will be fixed in next version. There is a workaround if you may want to continue to use Arduino version https://github.com/omersiar/esp-rfid/issues/39#issuecomment-355813168

Beardy241, 2018-02-06:
Thanks.  I have finally had time to look at it.

I can access the editor and have 'checked the config.json file' — but to be honest I don't know what I'm checking for.  It has my wifi credentials etc so it seems it has been set up successfully.  

I don't see a "latestlog.json" file which goermezer talks about.

commit 08df4af4 (2018-02-14): this should fix #58
closed 2018-02-14 by omersiar