omersiar/esp-rfid #59: Using Visual Studio Code with PlatformIO

akeilox, 2018-01-24:
I tried to run the latest arduino release on a previously working arduino install, which worked couple months ago, but kept getting esp board errors.

Taking that as an excuse, I have tried to download PlatformIO with Visual Studio Code and sharing my experience to get the latest branch to work on these combo;

Please note I have not used PlatformIO before and only seen the discussion on moving to this platform.
Therefore this might help others with similar upgrade path to get esp rfid running on platform io under windows 10 x64.

For Windows 10 x64;
Once you download the latest esprfid dev and create new project in VS Code PlatformIO, under platform.ini change the upload_port to  COMx where x is your com port. It was assuming Linux in esprfid dev version - as of this message.
Also when tried to upload the compiled code it was giving error on downloading tool_mkspiffs

So I downloaded this manually via https://bintray.com/package/files/platformio/dl-packages/tool-mkspiffs
for windows, version 2, and manually moved under directory c:\users\[username]\.platformio\packages\tool-mkspiffs

After which uploading the firmware was success. But the default build doesn't upload the spiff, 
so I ran the command pio run --target uploadfs
which uploaded the spiff as expected.

This is much better than fiddling with arduino ide library and board manager. The difference is day and night.

One point though, when I tried to use the Serial monitor in platformIO VScode, like in Arduino IDE, default 9600 baud, setting the com port correctly i was getting gibberish and couldnt find how to change the baud rate or get the serial output to work. Likely missing something here.
If anyone knows how to get it to work please kindly share here.

Quick observations on the dev branch as of this message;
On mobile chrome on iOS, the rfid settings page changes were not being shown after save reboot.
I had to switch to desktop chrome to get this working.

In desktop chrome, i observed the console is giving errors for retrieving woff2 and ttf fallback fonts ;
`
(Slow network is detected. Fallback font will be used while loading: http://192.168.1.135/fonts/glyphicons-halflings-regular.woff2
required.js:21 GET http://192.168.1.135/fonts/glyphicons-halflings-regular.woff2 net::ERR_ABORTED
getViewportWidth @ required.js:21
getWidth @ required.js:21
calculate @ required.js:21
get @ required.js:21
(anonymous) @ required.js:21
(anonymous) @ required.js:2
i @ required.js:2
add @ required.js:2
(anonymous) @ required.js:2
each @ required.js:2
(anonymous) @ required.js:2
Deferred @ required.js:2
then @ required.js:2
init @ required.js:21
(anonymous) @ required.js:21
Deferred @ required.js:2
_execute @ required.js:21
execute @ required.js:21
(anonymous) @ required.js:21
(anonymous) @ required.js:2
i @ required.js:2
add @ required.js:2
(anonymous) @ required.js:2
each @ required.js:2
(anonymous) @ required.js:2
Deferred @ required.js:2
then @ required.js:2
_init @ required.js:21
(anonymous) @ required.js:21
(anonymous) @ required.js:2
i @ required.js:2
add @ required.js:2
(anonymous) @ required.js:2
each @ required.js:2
(anonymous) @ required.js:2
Deferred @ required.js:2
then @ required.js:2
_construct @ required.js:21
(anonymous) @ required.js:23
(anonymous) @ required.js:21
construct @ required.js:21
k @ required.js:21
b.init @ required.js:21
(anonymous) @ users.js:42
i @ required.js:2
add @ required.js:2
n.fn.ready @ required.js:2
n.fn.init @ required.js:2
n @ required.js:2
initTable @ users.js:38
socketMessageListener @ users.js:275
users.htm:1 Slow network is detected. Fallback font will be used while loading: http://192.168.1.135/fonts/glyphicons-halflings-regular.woff
/fonts/glyphicons-halflings-regular.woff:1 GET http://192.168.1.135/fonts/glyphicons-halflings-regular.woff net::ERR_ABORTED
users.htm:1 Slow network is detected. Fallback font will be used while loading: http://192.168.1.135/fonts/glyphicons-halflings-regular.ttf
/fonts/glyphicons-halflings-regular.ttf:1 GET http://192.168.1.135/fonts/glyphicons-halflings-regular.ttf net::ERR_ABORTED)

`

omersiar, 2018-01-24:
Thank you for sharing your findings. Last code were had a hard coded font naming, this must be replaced by default css files. This is why its failing, also you have a workaround that might work, replace required.css files with the released old arduino code (using SPIFFS editor).

akeilox, 2018-01-24:
Thank you for the pointer. By SPIFF editor do you mean the rfidIP/edit editor? Cause the default platform io build it shows required.css.gz so editor is not loading the gunzipped css file.

Also I wanted to give MQTT and Wiegand a try (wiegand reader on its way), the default build only builds the nodemcu and skips nodemcu-mqtt and d2mini.
Assuming there is some setting wizardry to get these built, does it contain both the MQTT and Wiegand or just MFRC  + MQTT?

omersiar, 2018-01-24:
It will support both, yes goto /edit and delete the css.gz file and upload the old one which is bundled with release.

akeilox, 2018-01-24:
thanks!
i have changed the env_default to nodemcuv2_mqtt and built the mqtt version.
However, even though under [common] upload_port = COM6 is defined
the command pio run --target uploadfs tried to upload to COM5 (first one it finds). I dont understand why its not honoring the common setting anymore. If i use pio run --target uploadfs --upload-port COM6 then it works.

akeilox, 2018-01-24:
after uploading the .pioenvs\nodemcuv2_mqtt\firmware.bin and its spiff to esp, under setting page mqtt settings section is not showing up. Am I missing further setting in the platformio.ini beside changing env_default = nodemcuv2_mqtt

I have checked the https://github.com/omersiar/esp-rfid/blob/mqtt/CHANGELOG.md
and the sshot from the mqtt branch for where to expect mqtt settings but its not in the pio built version.

omersiar, 2018-01-25:
I believe MQTT never made into Platformio compatible code base. You should use plain old Arduino IDE to compile and upload.

closed 2018-02-14 by omersiar