omersiar/esp-rfid #60: In case of interrupt, log can be damaged.

nardev, 2018-01-31:
I have one client which was not able to retrieve log. When i downloaded the latestlof json file, i noticed that the log file was actually interrupted so the format was damaged.

If possible, please others test this case.. once the card is swiped and reader is interrupted, does anything wrong happen.

I hope i'm wrong.

goermezer, 2018-02-01:
@nardev : I had the same issue. Could solve it by removing the log file. Hole thread here: https://github.com/omersiar/esp-rfid/issues/39

nardev, 2018-02-01:
@goermezer @omersiar i'll try to test another way of writing the log..

Some clients i have, need that so it's not good if the log disappears..

Also, most of wiegand readers have keypad so i'll write and extensio that each user can have pin too. As an optional way to open the doors.

akeilox, 2018-02-02:
@nardev for wiegand readers with keypad, does it work if you enter card's decimal id manually?
Do you have to press a certain key after like # or other key after entering id?

just to chip in to the discussion, while trying out the wiegand reader with esp-rfid, if the 12V connection broken or re-established to the wiegand reader, the log file will capture single digit like 0 and next line 7.
Not something you'll normally encounter but I did recall the 'log file corrupted' discussions when i first saw it.

omersiar, 2018-02-02:
Logging is horribly inefficient, and also has a bug on file checking. @rneurink's approach was better and i forgot to fix my code.

At that time we were discussing different approaches for logging but it did not concluded, so if you make suggestions, we can start to work on it. Just do not bother with the current code.

nardev, 2018-02-02:
@akeilox  there is my board which can handle 12V for ESP and pass 12V to a reader.
Check it: https://www.tindie.com/products/nardev/esp-rfid-board-for-wiegand-readers-at-5v-to-28v/

If you want i can send you few without ESP12 module.. i have some that have a mistake for a reset button but everything else works.

akeilox, 2018-02-05:
@nardev thanks for the offer! i dropped you a message at tindie

omersiar, 2018-03-11:
See v6.0

omersiar, 2018-03-11:
See v0.6

closed 2018-03-11 by omersiar