omersiar/esp-rfid #61: [doc] Missing hint about security flaws

Rotzbua, 2018-02-11:
Hi,
nice project.

You should add to your readme that people should not only use this for access control because the uid of mifare cards can be easily cloned. So it is required to have a second access control instance like a normal key.

Reference: https://github.com/miguelbalboa/rfid#security

Best

nardev, 2018-02-12:
IMHO that is commonly know issue.

It also depends on device you use. In 99% of cases, the relay for doors is in the reader so, just removing the reader cover would give you opportunity to open the doors :D

omersiar, 2018-02-12:
Hi @Rotzbua,

It's nice to hear from you. We definitely need to be more specific about security of this project. Since the project is not yet a complete tool for access control we are open for more code to secure things up.

Kind Regards.

closed 2018-02-13 by omersiar