omersiar/esp-rfid #63: Restructure main.cpp as standard C++ project

omersiar, 2018-02-18:
It's now hard to maintain the project as its growing in size.

@chaintip

omersiar, 2018-02-18:
@chaintip 15 days

chaintip, 2018-02-18:
`0.00697993 BCH`| `~  10.61 USD` has [been tipped](https://explorer.bitcoin.com/bch/address/bitcoincash:qqh0ld4yzszqjet9f3s457sj2qsvc0tcfyk9ndl5k0) to this issue by @omersiar.
To claim [this bounty](http://github.chaintip.org/chaintip/bounties/issues/9), get a pull request merged with ```@chaintip fixes #63``` in the creation comment.

chaintip, 2018-03-20:
`0 BCH`| `~   0.00 USD` has now been tipped to this issue.
To claim [this bounty](http://github.chaintip.org/chaintip/bounties/issues/9), get a pull request merged with ```@chaintip fixes #63``` in the creation comment.

closed 2018-03-22 by omersiar