omersiar/esp-rfid #64: data dir missing

nardev, 2018-02-23:
scripts should prepare data dir for pio  to know from where to upload the data to spiffs

omersiar, 2018-02-24:
I think there will be no data dir at all.

Gulp script prepares web files for flashing to firmware, spiffs is no longer be used for web files anymore.

This is good because, we do not need to care about updating spiffs, user data can be preserved across firmware updates, etc

I did not test the behaviour of firmware update at this time. 

I will try to code web ui from ground up.

omersiar, 2018-02-24:
- [ ] Test firmware update beyond 512kb

nardev, 2018-02-28:
Tonight...

omersiar, 2018-03-04:
Testing is not a high priority task. I think with an optimization in future we can still use OTA update, also updating from SPIFFS is an option too. So let's see what future holds.

closed 2018-03-04 by omersiar