omersiar/esp-rfid #65: Latestspiffs for wiegand reader of last arduino release

JuculianD, 2018-02-27:
Hello, I have a problem with the spiff files for the last arduino version.
The option for the wiegand reader is missing in the settings.
[image]

I am trying to use that wiegand reader. It has an ST microcontroller on it.
Does it work?

Thanks for the development.

BTW, when will the new bootstrap web interface be ready?

Julian.

omersiar, 2018-02-27:
Hello @JuculianD,

The wiegand setting is not missing, it was not there at all in that release. So you need to wait for new release but i can not going to release it any time soon.

If your wiegand module with ST MCU is compatible with the protocol itself it should work as a reader.

JuculianD, 2018-02-27:
Thank you very much. I will wait for it. When approximately will it be ready?
I am building a codelock system for a school.
 Thank you for that development and license.

nardev, 2018-02-28:
@JuculianD ther is an old release which supports wiegand readers, have it working in some places. Also check this board https://www.tindie.com/products/edit/esp-rfid-board-for-wiegand-readers-at-5v-to-12v/

I made it specially for wiegand readers. 

Get this commit for wiegand readers: https://github.com/omersiar/esp-rfid/tree/f098df91b01ec2b9158f621790c12e24ea34932f

omersiar, 2018-02-28:
@JuculianD Hello,

I know you are waiting for it, you can build stable branch, it should have Wiegand reader support (i'm not sure tbh).

JuculianD, 2018-02-28:
> Thanks Oemer for the help! Unfortunately, your tindie link doesn't work.

@nardev
Thanks for the tindie link. I got it working. The boards are looking fine.
Are they preflashed with the firmware? How does the table data
synchronization work.

2018-02-28 15:53 GMT+00:00 Julian Dahl <vesteckt@gmail.com>:

JuculianD, 2018-02-28:
Thanks Oemer for the help! Unfortunately, your tindie link doesn't work.

nardev, 2018-02-28:
@JuculianD check again :D sorry https://www.tindie.com/products/nardev/esp-rfid-board-for-wiegand-readers-at-5v-to-12v/

nardev, 2018-02-28:
This is 50% promo code for the board:  F468260E

 https://www.tindie.com/products/nardev/esp-rfid-board-for-wiegand-readers-at-5v-to-12v/

JuculianD, 2018-03-02:
I need help... I have the 7941E v3.0 Reader but I cant read out anything. The reader has an UART and Wiegand option. I dont have any more information. Does somebody have an Idea?

JuculianD, 2018-03-02:
[image]
[image]

nardev, 2018-03-02:
You must use both wires for wiegand. That is the idea about the protocol. So hook D0 and D1, both to ESP. And determine which GPIO's are corresponding to that wemos, set it like that and there you go.

JuculianD, 2018-03-02:
I have both connected!

nardev, 2018-03-02:
Ou  see now.. at first i saw only white one... well, did you determine which GPIOs are corresponding to wemos pins?

JuculianD, 2018-03-02:
Yes, I put them in the settings page. I tried different pins as well, and I also tried a 3A 5V power supply now.

What is wrong? Is the reader damaged?

nardev, 2018-03-02:
@JuculianD so you use D7 and D8 which is GPIO 13 and GPIO 15.

And you used the commit that i suggested?

https://github.com/omersiar/esp-rfid/tree/f098df91b01ec2b9158f621790c12e24ea34932f

And you uploaded the spiffs files?

Am i right?

JuculianD, 2018-03-02:
I am using Pin D6 for D0 and 
D7 for D1.

JuculianD, 2018-03-02:
And no, I am not using your commit, because I does say invalid password for the login.

JuculianD, 2018-03-02:
And failed to open config file. I dont know why, but my wemos d1 mini is struggling with some firmwares?

nardev, 2018-03-02:
It's ok if you don't have config file, it will be recreated.

I don't know about new commits but the one i gave you, works 100%. If you have time, try that one.

JuculianD, 2018-03-02:
The new commit works fine, except of wiegand. Your commit doesn't work, I can't login.

nardev, 2018-03-02:
admin is the password

closed 2018-03-04 by omersiar