omersiar/esp-rfid #75: Scan of wifi returns more columns of the same wifi

romanzava, 2018-03-08:
Hi,

when I do a scan of wifi i see there are more times the same wifi networks.
[image]

SamNitro, 2018-03-08:
my scan is correct.
[image]

romanzava, 2018-03-08:
are some networks on N standard ? Because I think that it is messing it up a little.

SamNitro, 2018-03-08:
my PC network is N standard:

[image]

omersiar, 2018-03-09:
This is because of the recent limit. Loop does not take account number of the networks found, this is why you always see 5 networks, if there are 5 or more networks, this is not a problem.

@romanzava thank you.

omersiar, 2018-03-09:
Fixed with d873042

closed 2018-03-09 by omersiar