omersiar/esp-rfid #81: UART Wiegand reader

JuculianD, 2018-03-14:
Hello,
I found something out about the Wiegand reader that I am using and that - unfortunately - didn't work.

Same as in the Arduino forum thread, the problem is that the reader is connecting via UART.
Can someone help?

http://forum.arduino.cc/index.php?topic=152766.0

omersiar, 2018-03-14:
Do you want to use your Wiegand reader with esp-rfid? then you should find a wiegand library which supports uart and connect it to ESP8266's second UART.

macsjr, 2018-03-15:
The ESP8266 doesn't have a complete second UART, there is only TX on second UART.
Take a look at: http://esp8266.github.io/Arduino/versions/2.3.0/doc/reference.html#serial
To make this project work with my PN532 via UART I had to disable the debug messages sent via Serial.
Then I can use the serial port for communication with the nfc module.
Maybe it also works using SoftwareSerial, I didn't have the time to test yet.

omersiar, 2018-03-30:
Maybe we can send debug messages through UART1, and leave UART0 for RFID readers, since we do not listen data from serial.

macsjr, 2018-04-02:
UART1 is not connected to USB AFAIK if we go this route, people will need a uart to usb converter.
Maybe we can use Serial.swap()? This command changes TX0 and RX0 from GPIO1 and GPIO3 to GPIO15 and GPIO13.
We conect the RFID reader to GPIO15 and GPIO13. Everry time we need to comunicate to RFID reader we do a swap() before and swap() after. Things will get a bit messy, but it's doable.

omersiar, 2018-04-13:
I believe this can be done at a later time, but i can not make promises. Closing this issue for now.

closed 2018-04-13 by omersiar

ReneHezser, 2018-12-29:
My long range RFID reader is working in UART mode as well. Any chance it will be supported in the future?
I can read the uuid of my cards with the software serial from Arduino.