omersiar/esp-rfid #82: dual reader support

nedoskiv, 2018-03-14:
It will be nice to make it work with 2 readers to be able to put one on each side of the door. I have already done it in LUA:

https://github.com/nedoskiv/RC-RFID

akeilox, 2018-03-15:
i think it was discussed previously, it would be nice indeed if 2 readers supported with direction like ENTER and EXIT in Logging. Just saw your project in LUA script, is there any speed/memory difference of compiled code with LUA? just curious on the benefits of using LUA...

nedoskiv, 2018-03-15:
well actually there no benefit except that  I use known language to me. Guess arduino version can achieve better results , but since there is no project that meet my requirements i made that one and for now it covers most of what I need.

nedoskiv, 2018-03-15:
They are on same bus, SS(SDA) control which reader is used. Not tested cable length yet (on Orange PI was able to use 10M over FTP, here I need to rebuild everything since my prototype falls several times from the desk 
lol

omersiar, 2018-03-15:
Second reader makes harder to build a proper system. Maybe one can build as many as they want and make them communicate with each other over WiFi. Once I started to prototype it but i did not have the second reader and got bored.

akeilox, 2018-03-15:
i believe the second reader integration was an easy thought at the time; https://github.com/omersiar/esp-rfid/issues/7

where we only had MFRC522, and a matter of using additional SS pin to read one of the 2 cards since RST was dropped.
I really loved the idea of simple Enter and Exit rfid running on same esp-rfid, to simulate one door as one esp-rfid node like SideDoor. The logging of Enter and Exit could be stored as 1 or 0 to indicate direction.

Now with Wiegand and potential other readers it gets more complexity in implementation. 
But perhaps initial mfrc522 only enter/exit can be added first since the board is very cost effective, $2, and wiring works (i have a meter long cat5 cable connected to mfrc522, and it works).

nedoskiv, 2018-03-15:
Already done that, but in LUA, ESP send RFID, from which reader card is read, sensors info (if any attached) to an external server, it check  and return what it should do (switch relay, beep, switch lights) but it is a part from another project, not from STANDALONE RFID :)