omersiar/esp-rfid #83: Nodemcu Lolin board factory reset pin compatibility

macsjr, 2018-03-15:
Don't know if this is exactly an issue, but on the esp8266 board i'm using (lolin v3), the default fresetpin (16) can't be set to input pull up. Therefore every time the board is powered on the spiffs is formatted.
Maybe it can be changed to pin 13?

omersiar, 2018-03-16:
Thank you for feedback.

We are getting out of pins. It's hard to find one. I realized that GPIO-16 does not have PULL_UP resistor, no wonder why it's need to be connect to 3.3v. So, let me check options. I am open suggestions, pin 13 used by SPI so it's not going to work.

macsjr, 2018-03-16:
Thank you for this wonderful project, learned a lot with it.
Anyway, maybe this should be warned in the project description, to do a manual pull up (put a resitor 3v3->D0) and then enable/disable it in the web setup?
Maybe invert the logic, GPIO16 has internal pull-down (source: https://tttapa.github.io/ESP8266/Chap04%20-%20Microcontroller.html), so if it is HIGH, format.

omersiar, 2018-03-22:
Dropped reset by pin

closed 2018-03-23 by omersiar