omersiar/esp-rfid #84: Can't change config

fjfricke, 2018-03-19:
Hello together

Thanks for developing that wonderful software. Unfortunately I cannot change any settings as the following is sent over the serial connection:

> [ WARN ] Failed to open config file
> [INFO ] ESP-RFID is running in Fallback AP Mode
> [INFO ] Configuring access point... Ready

What is the problem there and how can I solve it?

omersiar, 2018-03-19:
The problem is that you need to connect gpio 16 to 3.3v

closed 2018-03-22 by omersiar

alegiovannelli, 2018-08-14:
Today i flashed the new version but i can't config some details in 'hardware settings' like imege attached.
Some combo box are locked.

Thank you in advance.
[image]

omersiar, 2018-08-15:
You flashed the v2board firmware binary, unless you have the official esp-rfid board you should have flashed generic.bin file.

alegiovannelli, 2018-08-15:
I did not notice the error during the flash, but by repeating it and paying attention I can now configure the hardware.
Thank you, wonderful project!