omersiar/esp-rfid #85: open / close with only one Reader

appi1, 2018-03-20:
Hello
this is an impressive project. I will use it for my camper, so I do not need a key (on the beach the key is lost).
Would it be possible, if I hold the RFID tag only briefly on the Reder, to close the motorhome (relay 1) and if I hold the RFID tag for more than 3 seconds to the Reder to open the motorhome (relay 2)?

Is there an idea for this issue?
Remo

omersiar, 2018-03-20:
Can't you use just one relay, why do you have seperate relays?

appi1, 2018-03-20:
Hello
in my motorhome, the closing mechanism is provided. a push-button to lock the vehicle and another button to unlock the car (Fiat Ducato).

Remo

JuculianD, 2018-03-20:
Why not just make a button for closing the vehicle. You don't need any security for the close button, do you??

appi1, 2018-03-21:
your are right, great Idea. thanks

closed 2018-03-22 by omersiar