omersiar/esp-rfid #86: Globilization

JuculianD, 2018-03-20:
I would like to translate into German. I would be happy if there is something like a language/translation file.

omersiar, 2018-03-22:
I am now considering best approach for globalization.

wejder669, 2018-03-29:
Hello All,
Could translate into Polish.
How many values are there to translate?

omersiar, 2018-04-13:
I'm closing this because there is no easy way to translate the whole project, unless we refactor the WebUI.  It will take time and I prematurely set it to 0.8 milestone and realized that it will take longer later.  So do not expect globalization at this time.

closed 2018-04-13 by omersiar

PierfrancescoElia, 2018-08-31:
I'm available to translate into Italian.
Feel free to contact when ready.

Thank you.

roysbike, 2018-09-09:
I can translate into Russian.