omersiar/esp-rfid #87: MQTT

nbaglietto, 2018-03-22:
How could MQTT be used? Is there an example?

Thank you
(I do not speak English)

SamNitro, 2018-03-24:
You need a MQTT broker first, like http://mosquitto.org 
My system run on Raspberry Pi

than you can publish the UID to the broker and subscribe it with another application like NodeRed

nbaglietto, 2018-03-24:
yo tengo instaldo mosquito en win7 lo que entiendo es que puedo hacer con el, recien instale node red en win y voy a probar pero quisiera alguna ayuda

SamNitro, 2018-03-24:
sorry with win 7 I'm out
-I use Raspberri Pi
-NodeRed
-FHEM

nbaglietto, 2018-03-25:
ok entonces supongamos que instalo mosquito y node-js en una vm con ubuntu 16 como se comunica con wemos d1 mini?

omersiar, 2018-03-25:
@SamNitro Did you have chance to test 0.7's MQTT functionality ?

SamNitro, 2018-03-25:
@omersiar Yes I have ;) and it works :)

@nbaglietto yes it should be work! If you have a MQTT Broker on ubuntu 16, you can read the Topic (message / UID) on win 7 (if win7 have a MQTT client)
The broker have to run 24/7 Hours, that the message can be receive.

rradar, 2018-03-27:
Yea! Now my esp-rfid works and I wonder what messages are published via MQTT?

Is there any overview somewhere? I have a home-assistant with mosqitto running and wonder what nice thinks I could do now with esp-rfid...

Thinking about putting a nfc-tag on my LP-cover and instead of playing the LP with the player I could just take the cover and touch the esp-rfid. It triggers the mqtt and home assistant knows to trigger mpd which takes the right album (from the uid) and just play's it...

thifi, 2018-03-27:
Everybody tested MQTT with CloudMQTT online? I did not get any message there.
Everytime I reboot the ESP the option for MQTT is disabled. 
Version is 0.7 on NodeMCU with RC522 (works correct)
Best regards

omersiar, 2018-03-27:
@rradar if enabled esp-rfid only publishes scanned rfid tag's uid on given topic. MQTT support is newly changed and we need testers. Feedbacks are welcome.

@thifi Does CloudMQTT supports unencrypted connections? It was working before?

thifi, 2018-03-27:
CloudMQTT support 3 different cennections like:
Port | 19977
SSL Port | 29977
Websockets Port (TLS only) | 39977
I used the "normal" port, I think this is unencrypted.
I used it before with ESP-Easy and MQTT with success.

omersiar, 2018-03-27:
@thifi I do not know how to debug MQTT issues. I do not have MQTT testing environment yet.

nbaglietto, 2018-03-27:
Hello, what I do not get is that esp-rfid is connected to my mosquito broker that I have in an ubuntu 16 in esp-rfid fill the subject field I put username and password and the ip address of the broque check the enable button but it does not connect and every time I renicio appears me the disabled mqtt.

(I do not speak English )

thifi, 2018-03-28:
@omersiar 
I get this message in the log-file of CloudMQTT if I try to connect from the esp-rfid:
2018-03-28 07:13:08: New client connected from 178.0.8.10 as ws0.8792021369477999 (c1, k60, u'pkvaifbf').
2018-03-28 07:14:57: Socket error on client ws0.8792021369477999, disconnecting.
I hope you can use it for the search of the problem...

pidiet, 2018-03-28:
I get feedback!

Settings in esp-rfid:

- local mqtt ip adress
- local mqtt port
- topc: "**/esp-rfid**"

And a local mosquitto server esp-rfid connects to (without user & password for testing)

To check my success I use mqttbox which looks like this:

[image]

pidiet, 2018-03-28:
@thifi 

maybe it's a problem with websocket? I tested with mqtt/tcp (without websocket/without encryption/without user+password) and it worked right away...

omersiar, 2018-03-29:
Can you flash this and look for event log? https://github.com/omersiar/esp-rfid/blob/dev/bin/firmware.bin

thifi, 2018-03-29:
Hi,
restart after flash ok, when I use MQTT data and restart: Exception (29) like on the picture
[image]

This is a serial copy, I could not get a webpage after restart with MQTT data.

omersiar, 2018-03-29:
[firmware.zip](https://github.com/omersiar/esp-rfid/files/1859164/firmware.zip)
Try this one please.

omersiar, 2018-03-29:
Clear event log and restart.

thifi, 2018-03-29:
Clear is ok, after restart the same problem with Please wait while fetching data...
After clear event log and access log there is no RFID Tag readable, no reaction, after reboot also.

omersiar, 2018-03-29:
[firmware2.zip](https://github.com/omersiar/esp-rfid/files/1859658/firmware2.zip)
:) Can you test this one, this should create a event log with a reason

thifi, 2018-03-29:
No eventlog at all, I don't whats wrong.
The RFID-Reading doesn't work.

thifi, 2018-03-29:
I think I have a hardware problem with 522, with original firmware 0.7 no RFID-reading :-)

thifi, 2018-03-29:
ok, no hardware problem, software. 
I flashed esp-easy, this formated the memory, then I flashed firmware2,
the esp started but no event-log.
The other points works, ntp, rfid, dhcp, user and access-log ok.

omersiar, 2018-03-30:
@thifi I couldn't reproduce this issue.  I do not understand why your MQTT service complains about websocket. MQTT simply tries to connect given port via TCP socket.

omersiar, 2018-03-30:
@SamNitro Hello,

Can you please test latest dev branch MQTT functionality.  To see if events correctly populated? Thank you.

SamNitro, 2018-03-30:
Hello, I was able to test MQTT successfully on the fast, but the event log does not show any data

"Please wait while fetching data..."

blmhemu, 2018-03-30:
I tried to establish MQTT connection in local host, as well as in cloud broker. None of them worked. I have checked the code where it was publishing and have serial printed something to see if the code is reaching there, I found that the code was actually reaching there. I think, the problem could be in the library. If you know some work around or need some help, please do tell. BTW this is a wonderful work you are doing here.

omersiar, 2018-03-31:
@blmhemu Thank you,

I built my MQTT test environment with mosquito. I am using RC522 reader. UID is publishing and event log works. Cloudmqtt somehow thinks esp tries to connect its Websocket and fails, sometimes Event log mentions MQTT_NOT_AUTHORIZED with cloudmqtt. I do not know why cloudmqtt is failing but esp definitely can not connect to it.  Also on mosquito _sub command lists publishes as like this `/esprfid/ 81a5428
` there is a space between / and uid, I do not know if this is normal behaviour.

Also it seems onMqttPublish does not work. maybe mosquito does not ack the pusblish.

magic-tune, 2018-03-31:
Hey omersiar. THX you do a great job!
Today I flashed the dev firmware. 
-upload the dev fw by using the webpage did not work for me. I got the same error as @thifi
- erasing the flash and reflash the dev fw did the trick.
-fix IP works fine
-mqtt enable/disable works fine
-mqtt (mosquitto with user/pw) doesn't work.
-mqtt (mosquitto without user/pw) works

omersiar, 2018-04-02:
Something fishy is going on between Async Web Server and Async MQTT.  Both depends on Async TCP and i believe there is a TCP client collusion somehow.

blmhemu, 2018-04-02:
@magic-tune 
I can confirm this. When I enter usr / pswd from web ui , it doesn't work.
But i did not try doing it from code.

magic-tune, 2018-04-03:
@omersiar I put the muser and mpas outside the scope. This works for me. If somebody wants to test. Here is the latest dev with my changes.
[firmware.zip](https://github.com/omersiar/esp-rfid/files/1872718/firmware.zip)

`
const char * muser;
const char * mpas;
{
String muserString = mqtt["user"];
muser = strdup(muserString.c_str());
String mpasString = mqtt["pswd"];
mpas = strdup(mpasString.c_str());
`