omersiar/esp-rfid #90: How to get 2/2MB Flash/SPIFFS Layout on a Wemos D1 Mini (4MB)

pidiet, 2018-03-25:
I tried hard... no way.. can't save anything...

Used arduino, esptool and google

Version [0.7.0]

> [firmware] !!! **BREAKING CHANGE** !!! 2 MB Flash 2 MB SPIFFS size for future proof firmware updates plase make sure you made a backup before updating to this version. You need to format SPIFFS.

omersiar, 2018-03-26:
Clean platformio directories and build again .it will automaticaly download git version of framework.

rradar, 2018-03-26:
I have the same problem when flashing the binary. Does it needed to be build from ground to overcome this "issue" or is there a trick that the binary can also work?

omersiar, 2018-03-26:
Factory format may help i guess.

rradar, 2018-03-26:
i did

> `esptool.py --port /dev/ttyUSB0 --baud 115200 erase_flash`

before flashing the binary. And the result is that I can't save any settings (guess it's releated to SPIFFS)

Any other command to try?

omersiar, 2018-03-26:
No i mean in WebUI there is option for factory reset.

myview01, 2018-03-26:
Hi, 
I tried everything (include factory reset), but stay in reboot loop.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

[ INFO ] ESP RFID v0.7
[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "",
    "ssid": "",
    "wmode": "",
    "pswd": "",
    "offtime": "0"
  },
  "hardware": {
    "readerType": "0",
    "wgd0pin": "4",
    "wgd1pin": "5",
    "sspin": "15",
    "rfidgain": "32",
    "rtype": "1",
    "rpin": "0",
    "rtime": ""
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": "0",
    "pswd": "admin",
    "version": ""
  },
  "mqtt": {
    "enabled": "0",
    "host": "",
    "port": "",
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": "30",
    "timezone": "0"
  }
}

Soft WDT reset

ctx: cont 
sp: 3fff1220 end: 3fff1590 offset: 01b0

>>>stack>>>
3fff13d0:  3fff0380 0023e72d 00000020 00000040  
3fff13e0:  3fff0380 3fff14b8 0000007d 4020b1fb  
3fff13f0:  3fff36b8 0000007d 3fff14b8 00000002  
3fff1400:  0000000f 3fff02d4 3fff0380 3fff3840  
3fff1410:  00000002 3fff02d4 3fff0380 40216003  
3fff1420:  00241001 00000001 3fff0570 40106ab9  
3fff1430:  00000002 3fff02d4 3fff0380 402162af  
3fff1440:  4020173e 00000001 00000032 40201733  
3fff1450:  0000000f 00000010 3fff02d4 4021641d  
3fff1460:  3fff36b8 00000020 3fff02d4 4021650d  
3fff1470:  0000000f 00000000 3fff02d4 4021652b  
3fff1480:  00000000 00000020 3fff3738 40234254  
3fff1490:  3ffeffb0 3fff049c 3fff3738 402358f8  
3fff14a0:  40107728 00000000 000003e8 4021c125  
3fff14b0:  3fff33bc 3fff33fc 3fff157d 3fff14e0  
3fff14c0:  4021c200 4021c358 40107310 001fb000  
3fff14d0:  3fff37bc 00000400 3fff14b8 00000292  
3fff14e0:  3fff049c 00001f20 3fff3414 4021a049  
3fff14f0:  3fff36d0 3fff37e8 3fff38b8 3fff055c  
3fff1500:  3fff17a8 3fff2698 3fff2694 4021c16f  
3fff1510:  3fffdad0 00000000 3fff049c 3fff055c  
3fff1520:  3fff17a8 3ffe94a5 3ffe90fc 40235c70  
3fff1530:  40208074 feefeffe 4021c66c 4021c658  
3fff1540:  00000000 00000000 feefeffe 00000000  
3fff1550:  00000000 feefeffe feefeffe feefeffe  
3fff1560:  feefeffe feefeffe feefeffe feefeffe  
3fff1570:  3fffdad0 00000000 3fff0555 4021b584  
3fff1580:  feefeffe feefeffe 3fff0570 40100721  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

omersiar, 2018-03-26:
@myview you need to connect rfid hardware to your esp. If you dont have any, choose wiegand on the reader type save and reboot.

myview01, 2018-03-26:
I have RC522 connected. I get to this state after select MFRC522 with GPIO15 -> Save -> Reboot.
HW - Wemos D1 mini - original (with ESPEASY works fine)
If I configure only Wifi connection, I get the similar state (reset loop) without stack dump.

pidiet, 2018-03-26:
I did a Factory rest but it doesn't help with the SPIFFS

omersiar, 2018-03-27:
@pidiet Please provide more information, are you working on Windows PC? 

@rradar Did you managed to solve the issue?

I successfully downgraded to 0.6.1 (via WebUI) formatted SPIFFs (via WebUI) (3MB usable), then updated to 0.7.0 formatted SPIFFs, saved settings, works good. I couldn't reproduce this issue.

Used binaries from release page. Did not erase the flash before, nothing i done in extra. just interacted with WebUI.

Used incognito mode on my mobile browser because of browser does not cache the pages in incognito mode.

Try erase.bat here to completely erase format the flash.

https://github.com/omersiar/esp-rfid/tree/dev/bin

I am out of ideas, beside the formatting the SPIFFs.

> [firmware] !!! BREAKING CHANGE !!! 2 MB Flash 2 MB SPIFFS size for future proof firmware updates plase make sure you made a backup before updating to this version. 

> You need to format SPIFFS.

omersiar, 2018-03-27:
@myview01 Did you managed to solve your issue? You may need to change gain, try different Gain settings for RC522.

rradar, 2018-03-27:
Hi @omersiar 
I also tried factory rest in esp-rfid after freshly flashing the 0.7.0 release on my wemos. But still I can't save any settings and the "FREE SPIFFS" has no color like from @pidiet 

I will try downgrading now. How exactly do you format the spiffs (only?)?

Thank's for your help!

rradar, 2018-03-27:
I downgraded (web ui) from 0.7.0 to 0.6.1 and got this dashboard:
[image]

after upgrading (web ui) to 0.7.0 again I have this one here:
[image]

omersiar, 2018-03-27:
@rradar did it work?

rradar, 2018-03-27:
@omersiar 
Yes! It's the first time the 'Free SPIFFS' is showing a green line. Before it was always grey. And now everything works like expected (I can finally save settings and active the reader -> works just out of the boy! great!)

So to sum' it up:
**It's necessary to flash a 0.6 release first and then upgrade via web ui to the newest 0.7 release.**

pidiet, 2018-03-27:
@omersiar 

I can agree the solution from @rradar . Flashing the old release and use the web ui for upgrade works!
Maybe would be good to point this out in the release section (just in case!)

Thank's a lot! I'm closing here...

closed 2018-03-27 by pidiet