omersiar/esp-rfid #91: Not working

teriall, 2018-03-27:
Hello! I tried to firmware before using Arduino IDE, when it was supported, now it not supported anymore. I tried to firmware the bin file, but only got this message in terminal: 
`ets Jan  8 2013,rst cause:2, boot mode:(3,7)
ets_main.c `
No wireless network appeared.
I'm using MFRC522, connected as described

omersiar, 2018-03-27:
Hi,

Which ESP module do you have? Which version did you flash? What OS are you on?

teriall, 2018-03-27:
my ESP module called Goouuu MINI S1, I think it's same with nodemcu. I flashing the latest one (0.7), flashing using flash.bat. On windows 10 x64

omersiar, 2018-03-27:
I guess your module's flash size or type is different, you can check it out with Arduino IDE example called CheckFlashConfig

Post the result here.

teriall, 2018-03-27:
`Flash real id:   001540E0
Flash real size: 2097152

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration wrong!`

omersiar, 2018-03-27:
It's working because back at Arduion times we did not care the flash size. As you can see your real flash size is 2MiB, you need 4MiB module in order to use esp-rfid fully. You can still modify it to use v0.6.1 or even latest version but you need to build it yourself.

Do you have another module or is this the only option you have?

omersiar, 2018-03-27:
Also pre-compiled binaries are for QIO mode, your flash works on DIO mode. I highly recommend getting ESP Module with 4MiB flash in QIO mode. Which is basically wemos D1 mini.

teriall, 2018-03-27:
Okay, understood! Thank you so much. I'll try to find the wemos d1 mini. Can close the issue:)

closed 2018-03-28 by omersiar

nikolaimann, 2018-04-01:
Hello, I have the same problem that my device does not start a wlan after flashing the precompiled binary.
The message is:
 ets Jan  8 2013,rst cause:3, boot mode:(3,6)
When executing the Arduino IDE example called CheckFlashConfig I get the following output:
```
Flash real id:   001640E0
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.
```
SO the flash size is ok, but it uses the wrong Flash IDE mode.
Is it possible to use the software with my module? What do I have to change?

teriall, 2018-04-02:
Actually I found a module which is 4Mb flash and DIO ide mode, but working just fine, without any problem. I checked the module schematic and compare to the wemos d1 mini schematic so the flash connected to the chip absolutely the same way.

omersiar, 2018-04-02:
@nikolaimann You need to edit platformio.ini and build firmware yourself. from `board_flash_mode = qio` to `board_flash_mode = dio`