omersiar/esp-rfid #93: Time does syn with ntp

pidiet, 2018-03-28:
Hello!

I have the problem that my time does not get synced with ntp. Not the stock settings nor my local ntp server are working. Devices always starts back in the year 1970....

The local sync with the browser works as expected but needs manual trigger.

[image]

omersiar, 2018-03-28:
It maybe related by DNS name resolution.

pidiet, 2018-03-29:
Really! It worked right away when giving the IP address of my local ntp server.

I think (my) esp-rfid is **not able to resolve host names only for ntp** (my mqtt is configured with host name and is working fine)

closed 2018-03-30 by omersiar