omersiar/esp-rfid #95: button to open from inside

teriall, 2018-03-29:
Is it possible to use one more pin for a button which will trigger the relay to open a door from inside?

omersiar, 2018-03-29:
You can connect button to relay circuit.

teriall, 2018-03-29:
ok:)

teriall, 2018-06-28:
I add the push button to trigger relay from the inside. Maybe possible to add in to the code? 
[image]

omersiar, 2018-07-01:
I think esp-rfid should not endorse additional hardware other than simply basic RFID reader support and a door lock trigger mechanism.  LCDs, Buzzers, Speakers, LEDs, Buttons does not fit for all, some are comfortable with Buzzer, some are implementing LCD.

We should keep it this way.

closed 2018-07-01 by omersiar

PierfrancescoElia, 2018-08-31:
Hi @teriall, I'm interested in adding a button.
Cloud you please share with the community your changes? 

I would really appreciate that! 
Thank you, have a nice day! :)

teriall, 2018-09-02:
I'll try to do it somehow