omersiar/esp-rfid #97: MFRC522 connected and still rebooting after update

skymike, 2018-04-01:
[ INFO ] ESP RFID v0.7
[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "",
    "ssid": "",
    "wmode": "0",
    "pswd": "",
    "offtime": "0"
  },
  "hardware": {
    "readerType": "0",
    "wgd0pin": "14",
    "wgd1pin": "12",
    "sspin": "15",
    "rfidgain": "112",
    "rtype": "0",
    "rpin": "5",
    "rtime": "1400"
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": "0",
    "pswd": "",
    "version": "v0.7"
  },
  "mqtt": {
    "enabled": "1",
    "host": "",
    "port": "1883",
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": "30",
    "timezone": "1"
  }
}

Exception (28):
epc1=0x4021e22b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1690 end: 3fff19e0 offset: 01a0

>>>stack>>>
3fff1830:  00000010 3fff18a8 3fff18a8 4021a0d7  
3fff1840:  3fff4074 00000004 3fff18a8 4021a123  
3fff1850:  3fff1c08 0000039a 0000039a 4010020c  
3fff1860:  3fff4074 3fff3bc0 3fff18a8 40208ebc  
3fff1870:  3fff4074 3fff3bc0 3fff0524 40209455  
3fff1880:  3fff3c28 3fff3bc0 3fff18a8 402350e7  
3fff1890:  401076c0 00000000 000003e8 00000008  
3fff18a0:  3fff3864 3fff38a4 00000000 00000000  
3fff18b0:  00000000 feefeffe 401072f8 00000020  
3fff18c0:  3fff3cac 00000400 3fff18a8 000002da  
3fff18d0:  3fff1910 3fff08f4 3fff2f20 3fff32e4  
3fff18e0:  3fff2494 3fff07f4 00000578 00000578  
3fff18f0:  000000f0 3fff2a28 3fff2a24 4021b7f1  
3fff1900:  00000578 4021cb34 3fff199c 4020c0e5  
3fff1910:  4021b968 4021ba24 11cf3d9c 001f1a9f  
3fff1920:  3fff38bc 00001000 00002000 00000100  
3fff1930:  3fff3907 3fff3d30 3fff3cd8 3fff3a07  
3fff1940:  0000001e 3fff2d44 00000000 3fff09b4  
3fff1950:  00000000 3fff2a28 3fff2a24 4021b83b  
3fff1960:  3fffdad0 00000000 3fff08f4 3fff09b4  
3fff1970:  3fff1bf8 3ffe971c 3ffe904c 4023547c  
3fff1980:  402081f4 feefeffe 4021bd88 4021bd74  
3fff1990:  00000000 00000000 feefeffe 00000000  
3fff19a0:  00000000 feefeffe feefeffe feefeffe  
3fff19b0:  feefeffe feefeffe feefeffe feefeffe  
3fff19c0:  3fffdad0 00000000 3fff09ad 4021ac58  
3fff19d0:  feefeffe feefeffe 3fff09c0 40100721  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

magic-tune, 2018-04-02:
Erase esp flash. You can do this with erase.bat in the bin folder. Then flash the 0.7 fw again.

omersiar, 2018-04-02:
It seems you do not enable MQTT but it is enabled by default, my mistake. Try to disable it before commit configuration. This will be fixed on in a couple of minutes on dev branch.

omersiar, 2018-04-02:
@magic-tune was right it's nothing to do with MQTT. See #99