omersiar/esp-rfid #98: NodeMCU V3 - Lolin and RC522

MrCrunsh, 2018-04-02:
My Setup:

NodeMCU V3 / LoLin 0.1
RFID-RC522

My Pin Setup

D1 -> GPIO5 -> SPI SDA
D2 -> GPIO4 -> Relay
D7-> GPIO 13 -> SPI MOSI
D6 -> GPIO 12 -> SPI MISO
D5 -> GPIO14 -> SPI SCK

I try to set SPI SDA on D8, but i will have the same issue :-(

Here ist my Serial output:

```
[ INFO ] ESP RFID v0.7
[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "[MyBSSID]",
    "ssid": "[MySSID]",
    "wmode": "0",
    "pswd": "[MyWifiPass]",
    "offtime": "0"
  },
  "hardware": {
    "readerType": "0",
    "wgd0pin": "4",
    "wgd1pin": "5",
    "sspin": "5",
    "rfidgain": "32",
    "rtype": "1",
    "rpin": "4",
    "rtime": "300"
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": "0",
    "pswd": "admin",
    "version": ""
  },
  "mqtt": {
    "enabled": "0",
    "host": "",
    "port": "",
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": "30",
    "timezone": "0"
  }
}

Exception (28):
epc1=0x4021e0e3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3fff12c0 end: 3fff1610 offset: 01a0

>>>stack>>>
3fff1460:  00000010 3fff14d8 3fff14d8 40219f47
3fff1470:  3fff40e4 00000005 3fff14d8 40219f93
3fff1480:  3fff1838 00000422 00000422 4010020c
3fff1490:  3fff40e4 3fff3c08 3fff14d8 40208d2c
3fff14a0:  3fff40e4 3fff3c08 3fff0154 402092c5
3fff14b0:  3fff3c70 3fff3c08 3fff14d8 40234f9f
3fff14c0:  401076d8 00000000 000003e8 00000008
3fff14d0:  3fff38bc 3fff38fc 00000000 00000000
3fff14e0:  00000000 feefeffe 40107310 00000020
3fff14f0:  3fff3cf4 00000400 3fff14d8 000002c5
3fff1500:  3fff1540 3fff0524 3fff3020 3fff333c
3fff1510:  3fff2fd4 4021b641 00000578 00000578
3fff1520:  000000f0 3fff2b98 3fff2b94 4021b669
3fff1530:  00000578 4021c9ec 3fff15cc 4020bf55
3fff1540:  4021b7e0 4021b89c 8b1ececc 001f1ed1
3fff1550:  3fff3914 00001000 00002000 00000100
3fff1560:  3fff3964 3fff3d78 3fff3d20 3fff3a4a
3fff1570:  0000001e 00000000 3fff0524 3fff05e4
3fff1580:  00000000 3fff2b98 3fff2b94 4021b6b3
3fff1590:  3fffdad0 00000000 3fff0524 3fff05e4
3fff15a0:  3fff1828 3ffe95c0 3ffe90fc 40235334
3fff15b0:  40208064 feefeffe 4021bc00 4021bbec
3fff15c0:  00000000 00000000 feefeffe 00000000
3fff15d0:  00000000 feefeffe feefeffe feefeffe
3fff15e0:  feefeffe feefeffe feefeffe feefeffe
3fff15f0:  3fffdad0 00000000 3fff05dd 4021aac8
3fff1600:  feefeffe feefeffe 3fff05f0 40100721
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

I do the follow one:
1. Git Pull
2. At `/bin` i used `erase.bat`
3. At `/ bin` i used `flash.bat`
4. Connect do AP ESP-RFID-XXXXX
5. Set up my Wifi Settings
6. Set up my Hardware Settings
7. Save all data

Then the issue goes on. Where is my fail?! Can anyone help me pls?

omersiar, 2018-04-02:
Can you try this version? https://github.com/omersiar/esp-rfid/releases/tag/0.6.1
Do not change any thing, first erase then flash 0.6.1 configure, save and reboot and report the result.

omersiar, 2018-04-02:
Please try latest version see #99

MrCrunsh, 2018-04-02:
Nice thank you! I will try it and give you a response later.

MrCrunsh, 2018-04-02:
I try latest version (Hotfix: #99) and now i get these bootloop:

```
[ INFO ] ESP RFID v0.7
[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "CC:CE:1E:8B:D1:1E",
    "ssid": "DerSchwarm",
    "wmode": "0",
    "pswd": "FamilieHering07072017",
    "offtime": "0"
  },
  "hardware": {
    "readerType": "0",
    "wgd0pin": "4",
    "wgd1pin": "5",
    "sspin": "5",
    "rfidgain": "32",
    "rtype": "1",
    "rpin": "4",
    "rtime": "300"
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": "0",
    "pswd": "admin",
    "version": ""
  },
  "mqtt": {
    "enabled": "0",
    "host": "",
    "port": "",
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": "30",
    "timezone": "0"
  }
}

Exception (28):
epc1=0x4021e12f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3fff12e0 end: 3fff1630 offset: 01a0

>>>stack>>>
3fff1480:  00000010 3fff14f8 3fff14f8 40219f93
3fff1490:  3fff4104 00000005 3fff14f8 40219fdf
3fff14a0:  3fff1858 00000422 00000422 4010020c
3fff14b0:  3fff4104 3fff3c28 3fff14f8 40208d2c
3fff14c0:  3fff4104 3fff3c28 3fff0174 402092c5
3fff14d0:  3fff3c90 3fff3c28 3fff14f8 40234feb
3fff14e0:  401076d8 00000000 000003e8 00000008
3fff14f0:  3fff38dc 3fff391c 00000000 00000000
3fff1500:  00000000 feefeffe 40107310 00000020
3fff1510:  3fff3d14 00000400 3fff14f8 000002c5
3fff1520:  3fff1560 3fff0544 3fff3020 3fff335c
3fff1530:  3fff2ff4 4021b68d 00000578 00000578
3fff1540:  000000f0 3fff2bb8 3fff2bb4 4021b6b5
3fff1550:  00000578 4021ca38 3fff15ec 4020bfa1
3fff1560:  4021b82c 4021b8e8 8b1ececc 001f1ed1
3fff1570:  3fff3934 00001000 00002000 00000100
3fff1580:  3fff3984 3fff3d98 3fff3d40 3fff3a6a
3fff1590:  0000001e 00000000 3fff0544 3fff0604
3fff15a0:  00000000 3fff2bb8 3fff2bb4 4021b6ff
3fff15b0:  3fffdad0 00000000 3fff0544 3fff0604
3fff15c0:  3fff1848 3ffe95e7 3ffe90fc 40235380
3fff15d0:  40208064 feefeffe 4021bc4c 4021bc38
3fff15e0:  00000000 00000000 feefeffe 00000000
3fff15f0:  00000000 feefeffe feefeffe feefeffe
3fff1600:  feefeffe feefeffe feefeffe feefeffe
3fff1610:  3fffdad0 00000000 3fff05fd 4021ab14
3fff1620:  feefeffe feefeffe 3fff0610 40100721
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```
I try to flash 0.6.1 now!

MrCrunsh, 2018-04-02:
At 0.6.1 the interface don't work. I cant scan any Wifis. I try to set up it manualy, but no data would be saved.

MrCrunsh, 2018-04-02:
Ok now i try it incognito mode and i got this issue:

´´´
[ INFO ] ESP RFID v0.7
[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "[MyBSSID]",
    "ssid": "[MySSID]",
    "wmode": "0",
    "hide": "0",
    "pswd": "[MyPass]",
    "offtime": "0",
    "dhcp": "1",
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": ""
  },
  "hardware": {
    "readerType": "0",
    "wgd0pin": "4",
    "wgd1pin": "5",
    "sspin": "5",
    "rfidgain": "32",
    "rtype": "1",
    "rpin": "4",
    "rtime": "300"
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": "0",
    "pswd": "admin",
    "version": ""
  },
  "mqtt": {
    "enabled": "0",
    "host": "",
    "port": "",
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": "30",
    "timezone": "0"
  }
}
[ INFO ] Trying to connect WiFi: [MySSID]........
[ INFO ] Client IP address: 192.168.2.40

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

MrCrunsh, 2018-04-02:
I use the Hotfix firmware from #99

omersiar, 2018-04-02:
Does your RC522 module properly connected to esp?

MrCrunsh, 2018-04-02:
Yes, it is. I have checked several times.

omersiar, 2018-04-02:
[firmware.zip](https://github.com/omersiar/esp-rfid/files/1868522/firmware.zip)
Can you please test this firmware?

MrCrunsh, 2018-04-02:
After flashing and set up my data, i get these:
```
[ WARN ] Failed to open config file
[ INFO ] ESP-RFID is running in Fallback AP Mode
[ INFO ] Configuring access point... Ready
[ INFO ] AP IP address: 192.168.4.1
[ INFO ] AP SSID: ESP-RFID-3474a1

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
AP closed and 2 sec later i see the ap again, but can't connect to it.

But the NodeMCU don't connect with my Wifi.

I reconect the voltage on NodeMCU and get these:

```
[ INFO ] ESP RFID v0.7
[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "[MyONe]",
    "ssid": "[MyONe]",
    "wmode": "0",
    "hide": "0",
    "pswd": "[MyONe]",
    "offtime": "0",
    "dhcp": "1",
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": ""
  },
  "hardware": {
    "readerType": "0",
    "wgd0pin": "4",
    "wgd1pin": "5",
    "sspin": "5",
    "rfidgain": "32",
    "rtype": "1",
    "rpin": "4",
    "rtime": "300"
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": "0",
    "pswd": "admin",
    "version": ""
  },
  "mqtt": {
    "enabled": "0",
    "host": "",
    "port": "",
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": "30",
    "timezone": "0"
  }
}
[ INFO ] Trying to setup RFID Hardware :[ INFO ] RFID SS_PIN: 5 and Gain Factor: 32
[ INFO ] MFRC522 Version: 0x92 = v2.0
[ INFO ] Trying to connect WiFi: [MyONe]........
[ INFO ] Client IP address: 192.168.2.40

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

omersiar, 2018-04-04:
@MrCrunsh Hello,

Are you still having this issue? I can prepare more debugging options for you to try.

MrCrunsh, 2018-04-04:
Yes i still having this issue... All trys goes in a loop...
I don't think it is a power supply issue, because i use a 2A power supply.

omersiar, 2018-04-04:
Can you please test 85.199.214.98 as ntp server?

MrCrunsh, 2018-04-04:
Still same issue...

MrCrunsh, 2018-04-04:
In version 0.6.1 the process goes through: Flash and then the board restarts and creates an AP. With version> 0.7 I get the following error after flashing and the board must be reset or removed from the power supply:
```
ets Jan  8 2013,rst cause:2, boot mode:(1,6)
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

MrCrunsh, 2018-04-04:
But after set up my data (Hardware and Wifi) and store them, i got the same loop as v0.7

omersiar, 2018-04-04:
Can you post results of CheckFlashConfig's example? Which is bundled with Arduino IDE.

```c++
/* 
 ESP8266 CheckFlashConfig by Markus Sattler
 
 This sketch tests if the EEPROM settings of the IDE match to the Hardware
 
 */

void setup(void) {
    Serial.begin(115200);
}

void loop() {

    uint32_t realSize = ESP.getFlashChipRealSize();
    uint32_t ideSize = ESP.getFlashChipSize();
    FlashMode_t ideMode = ESP.getFlashChipMode();

    Serial.printf("Flash real id:   %08X\n", ESP.getFlashChipId());
    Serial.printf("Flash real size: %u\n\n", realSize);

    Serial.printf("Flash ide  size: %u\n", ideSize);
    Serial.printf("Flash ide speed: %u\n", ESP.getFlashChipSpeed());
    Serial.printf("Flash ide mode:  %s\n", (ideMode == FM_QIO ? "QIO" : ideMode == FM_QOUT ? "QOUT" : ideMode == FM_DIO ? "DIO" : ideMode == FM_DOUT ? "DOUT" : "UNKNOWN"));

    if(ideSize != realSize) {
        Serial.println("Flash Chip configuration wrong!\n");
    } else {
        Serial.println("Flash Chip configuration ok.\n");
    }

    delay(5000);
}
```

omersiar, 2018-04-04:
[firmware.zip](https://github.com/omersiar/esp-rfid/files/1876063/firmware.zip)
Can you please try this one? Are you on a windows pc or other? How do you flash it? After flashing does it find configuration file? Try erasing before flash.

MrCrunsh, 2018-04-04:
I use a Windows PC. I flash with the tools in the / bin folder. Before I flash I use the erase.bat.

MrCrunsh, 2018-04-04:
I use a Windows PC. I flash with the tools in the `/bin` folder. Before I flash I use  `erase.bat`.

Now after flashing it starts self and i get this output:

```
[ INFO ] ESP RFID v0.7
Flash real id:   001640EF
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

[ WARN ] Failed to open config file
[ INFO ] ESP-RFID is running in Fallback AP Mode
[ INFO ] Configuring access point... Ready
[ INFO ] AP IP address: 192.168.4.1
[ INFO ] AP SSID: ESP-RFID-3474a1
```
My Configs:
```
{
  "command": "configfile",
  "network": {
    "bssid": "[MyBSSID]",
    "ssid": "[MySSID]",
    "wmode": 0,
    "hide": 0,
    "pswd": "[MyPass]",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": ""
  },
  "hardware": {
    "readerType": 0,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "sspin": 15,
    "rfidgain": 32,
    "rtype": 1,
    "rpin": 4,
    "rtime": 400
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "admin",
    "version": "v0.7.3"
  },
  "mqtt": {
    "enabled": 0,
    "host": "",
    "port": 1883,
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0
  }
}
```
Then i klick `save & reboot`

I see the `Please wait while your ESP restarts...` window but on my serial monitor nothing happend.

After restart my NodeMCU:
```
[ INFO ] ESP RFID v0.7
Flash real id:   001640EF
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

[ WARN ] Failed to open config file
[ INFO ] ESP-RFID is running in Fallback AP Mode
[ INFO ] Configuring access point... Ready
[ INFO ] AP IP address: 192.168.4.1
[ INFO ] AP SSID: ESP-RFID-3474a1
```
It seems like no config is stored

omersiar, 2018-04-04:
When you click save and reboot it should reboot at that moment, which browser currently trying on?

MrCrunsh, 2018-04-04:
It seems like the board reboot. I use Chrome.

MrCrunsh, 2018-04-10:
Could you reproduce my issue? 
Is there a solution?

omersiar, 2018-04-10:
I could not actually, it is working quite well on wemos d1 mini, i do not know what's failing. The major change is async mqtt library, you may want to use older versions of esp-rfid

MrCrunsh, 2018-04-12:
I have now bought a weemos D1 mini. Everything works perfectly here.
The same firmware and the same settings are running:

```
[ INFO ] ESP RFID v0.7
[ WARN ] Failed to open config file
[ INFO ] ESP-RFID is running in Fallback AP Mode
[ INFO ] Configuring access point... Ready
[ INFO ] AP IP address: 192.168.4.1
[ INFO ] AP SSID: ESP-RFID-33fd2f

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

[ INFO ] ESP RFID v0.7
[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "[MyBSSID]",
    "ssid": "[MySSID]",
    "wmode": "0",
    "hide": "0",
    "pswd": "[myPasswort]",
    "offtime": "0",
    "dhcp": "1",
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": ""
  },
  "hardware": {
    "readerType": "0",
    "wgd0pin": "4",
    "wgd1pin": "5",
    "sspin": "15",
    "rfidgain": "32",
    "rtype": "1",
    "rpin": "2",
    "rtime": "300"
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": "0",
    "pswd": "admin",
    "version": ""
  },
  "mqtt": {
    "enabled": "0",
    "host": "",
    "port": "",
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": "30",
    "timezone": "0"
  }
}
[ INFO ] Trying to connect WiFi: [MySSID].........
[ INFO ] Client IP address: 192.168.2.99
[ INFO ] Configuration done.
```

If I now load the same pin assignment, the same software and the same configuration on the NodeMCU LoLin, it does not work.

omersiar, 2018-04-13:
Since ESP8266 chip is quite stable, i think this maybe is about flash chip configuration.  Currently i do not have any other dev board other than Wemos and ESP-07 module.  Firmware works without an issue when configured right for each board. Now I assume this can be fixed easily but first we need to understand what causes it.

closed 2018-04-13 by omersiar